RavenSystem/esp-homekit-devices #569: flashing with esptool get - no buf for probe, ie len 0

vsidlu, 2019-10-28:
After flashing device I have this in log!

19:49:21.350 -> scandone
19:49:29.282 -> no buf for probe, ie len 0
19:49:29.428 -> no buf for probe, ie len 0
19:49:29.564 -> no buf for probe, ie len 0
19:49:29.707 -> no buf for probe, ie len 0
19:49:29.846 -> no buf for probe, ie len 0
19:49:31.350 -> scandone
19:49:41.366 -> scandone
19:49:49.409 -> no buf for probe, ie len 0
19:49:49.544 -> no buf for probe, ie len 0
19:49:49.718 -> no buf for probe, ie len 0
19:49:51.361 -> scandone

 can not connect to flashed device! What I did wrong?

RavenSystem, 2019-10-30:
Try erasing flash and flashing it again.

vsidlu, 2019-10-30:
did it! some times it helps, but some times it is still in this **no buf for probe, ie len 0** mode. flashed from two different macs and get same log. thought that you now what this log is about. will keep searching for reason. thanks for response!

closed 2019-10-30 by vsidlu