RavenSystem/esp-homekit-devices #570: HAA as motion sensor

sorriso93, 2019-10-28:
Hello
this is my configuration with a wemos d1 mini plus wemos PIR shield

{"c":{"l":2,"i":1},"a":[{"t":12,"f1":[{"g":0}],"b":[],"s":0,"i":5}]} 

The problem is that seems the sensor sometime is sleeping. To be clear, even passing the hand in front of the sensor sometime is fully responding (i.e. I see in home app the motion activated and the led on board blinking) sometimes it does not respond even passing the hand at distance of1 cm, fully randomly.
This behavior is not related to distance from the wifi signal of the sensor and board, verified the same at 1 m from the wifi hub.
Any idea?
Many thanks

sorriso93, 2019-11-01:
Thanks now It get constant behaviour but it activates after 3 sec I move the hand in front of on homekit, is it normale this delay?

RavenSystem, 2019-11-01:
No, I think your sensor works in the normal way. JSON should be:

{"c":{"l":2},"a":[{"t":12,"f1":[{"g":0}],"i":5}]}

sorriso93, 2019-11-02:
Now it’s ok, only note the firts time it gets motion delayed, maybe is the IR sensor. Waiting for the light sensor implementation 😂 wemos.cc for datasheet of the shield sensor. Many thanks

closed 2019-11-02 by sorriso93