RavenSystem/esp-homekit-devices #577: Feature request: prevent hardware buttons from changing GPIO output.

vsevolod-volkov, 2019-11-03:
Feature request goes from need to use Shelly 1 as interface between currently installed pulse relay (Legrand 4 124 12, Two pole latching relay - standard - 16 A 230 V - 2 N/O) and HomeKit environment.

I want to use my Shelly 1 switch as binary HomeKit switch sending short pulse to GPIO4 when using HomeKit button, while using GPIO5 as relay state indicator. Implementing this is impossible due to automatic action triggering on input change resulting multiple pulse relay state change.

So I am asking to implement in HAA special button feature for keeping it from changing GPIO output when it triggers action for HomeKit such as shown below as f0.a and f1.a:
```
{
    "a": [
        {
            "t": 1,
            "0": {
                "r": [
                    {
                        "g": 4,
                        "v": 1,
                        "i": 0.1
                    }
                ]
            },
            "1": { "a": 0 },
            "f0": [
                {
                    "a": 0,
                    "g": 5,
                    "t": 0
                }
            ],
            "f1": [
                {
                    "a": 0,
                    "g": 5,
                    "p": 0
                }
            ]
        }
    ]
}
```

RavenSystem, 2019-11-03:
I don't understand what you mean.

vsevolod-volkov, 2019-11-04:
Ok. I will try to explain.

Pulse relay - device that inverts its output state every time it gets short pulse at its input.
[image]
[image]

I have ten pulse relays overall on DIN-rails in my house and I want to control them with number of Shellys with HAA onboard. Pulse relays being controlled by wired switches are expensive components but it is very reliable, more-more reliable as cheap relay used in Shelly 1, so I do not want to change ones to anothers.

To implement that idea I need HAA binary switch ability to:
1) send short pulses to output GPIO (resulting pulse relay to invert its output state);
2) get actual switch via input GPIO (registering pulse relay output state).

Actually both those requirements are satisfied in current HAA release except one annoying thing: every time I use wired switch on the wall to on/off light, HAA detects input GPIO change then initiates pulse at output GPIO resulting changing light state again.

So, I would appreciate you for implementation in HAA the ability to disable applying output GPIO state when action is initiated by input GPIO, while keep output GPIO change  when action is initiated from HomeKit side.

RavenSystem, 2019-11-04:
Thanks! I understand now. But your request can not be done because internal code structure. Actions are controlled by HomeKit service, and digital inputs manage HomeKit service.

You must use a Shelly 1 with a custom firmware to perform it, connecting the L of the lamp to Shelly S1 connector, so you can read status of your lamp.

Duplicate issue: https://github.com/RavenSystem/esp-homekit-devices/issues/550

closed 2019-11-04 by RavenSystem

vsevolod-volkov, 2019-11-17:
I have an idea how to adopt bistable pulse relay to HomeKit and not to break HomeKit paradigm.
It is possible to implement condition for checking input GPIO before applying change to output GPIO like in example shown below:
```
            "0": {
                "r": [
                    {
                        "c": {
                          "g": 5,
                          "p": 0
                        },
                        "g": 4,
                        "v": 1,
                        "i": 0.1
                    }
                ]
            },
```
This example assumes the "c" (condition) element has the same subelement meaning as described for button.

This approach has possible additional benefit: consumer gets the ability to program some kind of "smart" actions like "set output GPIOn if input GPIOx is on and set output GPIOm, when input GPIOx is off".