RavenSystem/esp-homekit-devices #579: Sonoff Dual - HAA code

hexardavies, 2019-11-04:
Have had it working over the weekend. works great. try it.

{"c": {"l": 13,"b": [{"g": 10}]},"a": [{"0": {"r": [{"g": 5}]},"1": {"r": [{"g": 5,"v": 1}]},"b": [{"g": 0}]},{"0": {"r": [{"g": 12}]},"1": {"r": [{"g": 12,"v": 1}]},"b": [{"g": 9}]}]}

RavenSystem, 2019-11-07:
Many thanks.!

closed 2019-11-07 by RavenSystem

sercitecotf, 2019-11-12:
and in switch mode instead of switch, how would it be?

RavenSystem, 2019-11-15:
@sercitecotf 

`{"c":{"l":13,"b":[{"g":10}]},"a":[{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g": 0},{"g":0,"t":0}]},{"0":{"r": [{"g":12}]},"1": {"r":[{"g":12,"v":1}]},"b":[{"g":9},{"g":9,"t":0}]}]}`

But it is not recommended, because when gpio 0 is pushed, device will not boot because it will enter in flash mode. To use with 2 switches instead buttons, Shelly 2.5 is better.