RavenSystem/esp-homekit-devices #580: Doubts about the price? This project is TOTALLY FREE

RavenSystem, 2019-11-04:
This is just a reminder to those who still do not know that everything about this project is totally free.

If you see material from here in other webs, supported by ads, forced registrations, forced donations... remember that they are not supported by me. This is not allowed due to point 10 of LICENSE.

I created this project and firmwares for my house, and why not share them with all the world?

"Knowledge belongs to the world."
            Hackers 3 Movie (Anti Trust)

Bene2103, 2019-11-04:
AND we‘re thankful for your engagement! 

You need any device to test? Let me know!

And again, thank you very much! I‘m loving your firmware.

curla92, 2019-11-04:
Dear Josè, Our italian group never thought to speculate on your firmware. There was a misunderstanding,  we always appreciate your works and we are always available for a clarification. Our forum has been created to support your firmware and help the community to use it. The forum costs a little so we use ads to cover the cost only, we don't profit. I hope to understand. Thank y

sorriso93, 2019-11-04:
Well what I have to say is that we, italians, always have elastic interpretation of rules. I’m italian but I have to agree with Jose, sorry that you and the others think different, I’m with Jose and, with coherence, I leaved the italian group fin the meantime. Good luck, hope you’ll change idea.

curla92, 2019-11-04:
> Well what I have to say is that we, italians, always have elastic interpretation of rules. I’m italian but I have to agree with Jose, sorry that you and the others think different, I’m with Jose and, with coherence, I leaved the italian group in the meantime. Good luck, hope you’ll change idea.

Who decides the rules? We always support josè, we always put donate links in our group and always collaborate with him. If he had any dubs, he could speak with us to resolve the problems without create an useless scene. A few day ago we have been deleted from github page as ravencore support without notice. If josè doesn’t wont continue the collaboration it’s ok but we are not making profit because they are baseless accusations.

sorriso93, 2019-11-05:
Francesco the rules are by jose, simple. It his its work and we should always respect the work by people releasing it fo free (also you made great works). For sure there’s no profit from banners, I know perfectly, but if he defines the rules and want it to be respected that’s it. I disagree with what has been wrote after jose decision.

homekititalia, 2019-11-05:
I use the official channel of Homekit Italia as Mr. Josè blocked my account! (why?!?!) This is also proof of the seriousness of the person with whom we have worked for months. Perhaps Mr. Josè forgets the fact that with our group he received a lot of money! Money that arrived from Italian users! Then from one day to the next, without telling us anything, it removes us, cancels us, eliminates us. Not to mention the fact that with a fake profile he was spying us on Facebook. Well, we are not mafiosi and you are not an FBI agent! We are a community that has grown thanks to a diligent and hard work! We have thrown away hours and hours explaining to people how the firmware worked and how devices are flashed.
We have created websites and forums to discuss them, spending our money (and not a few)! All this without ever asking anyone for anything! Indeed, if we asked for something we would do it for you! We asked our users to make a donation! And with the donations of our users you have enriched yourself and not a little! Unlike us you have no expense! So please don't be a moralist. Look pathetic otherwise! All you did was done for you, for your home (I think it will be a castle given the length of the construction site). We did it for the community instead! So don't come and teach us anything! You are not in a position to do it! Instead of watching movies and quoting "Knowledge belongs to the world" phrases (I would like to tell you that this quote was also in the first part of the film) read some books and mature a little inwardly. Here in Italy it is said "you still have a mouth dirty with milk". Have a nice day :-)

curla92, 2019-11-05:
I repeat that it is just a misunderstanding and I am really sorry that there is no possibility of dialogue.

The matter ends here. I hope it is clear that we are not making a profit on your work.

sorriso93, 2019-11-05:
Guys what I can say is that both of you have done great work with development and guides and as a simple user I have to thank both of you. I think also that none of you became reach with donations or banners considering also that to do what you are doing a lot of free time is dedicated to this work. I hope that sooner or later it will be time to collaborate again together. Cheers

RavenSystem, 2019-11-05:
Due to the lies, menaces and abuses of [HomekitItalia group comment](https://github.com/RavenSystem/esp-homekit-devices/issues/580#issuecomment-549786098) about me and the work that this community realize, this group of people is not welcome to this community.

closed 2019-11-05 by RavenSystem