RavenSystem/esp-homekit-devices #586: HAA - Thermostat with sensor Si7021

OnKram, 2019-11-06:
Hi Jose,
I have trouble with set other sensor types on your template.
If I set the sensor type "n":4 (Si7021)  as the accessory type it doesn't work.
Could you help me pls.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":21,"n":4,"s":5,"g":14,"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0}}]}

bj-tracer, 2019-11-06:
Si7021 sensor have a I2C bus. It isn't a single-wire interface.

OnKram, 2019-11-06:
But we have four sensors to choose, so Si7021 should working.

bj-tracer, 2019-11-07:
I think it for Sonoff version of sensor. It has i2c to 1-wire converter.
https://www.itead.cc/wiki/Sonoff_Sensor_Si7021
So you can't use sensors from AliExpress with 4 pins (Vcc/SDA/SCL/GND).

RavenSystem, 2019-11-07:
Si7021 **one-wire** version is supported, but esp-open-rtos driver fails with many sensor like it. I recommend to use a DHT22 or DHT11 compatible sensor.

WizBangCrash, 2019-11-29:
HI @RavenSystem , can we re-open this issue as I have confirmed there is a bug in the esp-open-rtos code where the raw values returned by the Si7021 sensor are not being interpreted correctly.
I have submitted a pull request to the esp-open-rtos maintainers [#737](https://github.com/SuperHouse/esp-open-rtos/pull/737) and hopefully my fix will be merged soon.

RavenSystem, 2019-11-29:
Great! As soon as this is merged, I will release an updated.

WizBangCrash, 2019-11-29:
Thanks.
Question: I've built my own firmware and flashed it on a few devices to confirm the fix. I want to reassemble those device and put them back into operation in my home. If I go into setup mode and check the "Auto OTA Update" option once you have released a version that includes the fix. Will the device then update with the "official" firmware?
I'm trying to avoid having to retrieve the devices from the various places around the home and manually flash them to get them back on the "official" release.

mafyata, 2019-11-29:
Auto ota update is updating firmware when you enter setup mode but without connect to haa wifi.

RavenSystem, 2019-11-29:
@WizBangCrash are you able to see “auto ota update” when you enter in setup mode?

WizBangCrash, 2019-11-29:
@RavenSystem Yes. I can see three options with my modified version of the 0.7.4 firmware. See screenshot.
[image]

RavenSystem, 2019-11-29:
That is good, because you installed first a normal installation using OTA, and then you did not erase flash before installing your modified firmware. This means that you can update your device using OTA.

WizBangCrash, 2019-11-29:
Thanks for the feedback. I was hoping that would be the case .

Codkit, 2019-12-02:
Hi Jose

I also got the Sonoff Si7021 which is one wire data , 4 pins :
[https://www.itead.cc/wiki/images/8/80/Si7021_audio_connector.jpg
](url)
With TH16 only the switch is working,not the TH......

RavenSystem, 2019-12-04:
Try v0.7.9, please.

WizBangCrash, 2019-12-05:
Tested 0.7.9 on TH16+Sonoff AM2301 this evening and working fine. Thanks.
I'll test the Si7021 and ASAIR AM2301 at the weekend.

WizBangCrash, 2019-12-08:
Tested with 0.7.10 on TH16+Sonoff AM2301, TH10+ASAIR AM2301 & TH10+Sonoff Si7021.
All working fine, including using different temperature and humidity offsets.

RavenSystem, 2019-12-08:
Many thanks!

closed 2019-12-08 by RavenSystem

nepadm, 2020-12-01:
Please write correct JSON config for SONOF TH16 and Si7021 sensor.

Danilo0609, 2021-11-10:
Pueden escribir el JSON completo corregido?