RavenSystem/esp-homekit-devices #588: Si7021 sensors not working

s3phy, 2019-11-08:
Hello,

This is probably an upstream bug, but I haven't found out how to test my sensors directly with esp-open-rtos without HAA.

I bought several Si7021 sensors and I have this issue with all of them. (Can't seem to find AM2301 sensors anymore, last ones I bought I actually got Si7021 instead!)

The temperature always read 0°C and the humidity always reads 2%.

In the serial console, I see

HAA > TEMP 0, HUM 2
HAA > TEMP 0, HUM 2
HAA > TEMP 0, HUM 2
HAA > TEMP 0, HUM 2

If I choose another type of sensor, I get "HAA ! ERROR Sensor".

With tasmota firmware the sensors work correctly both with temperature and humidity.

Using HAA 0.6.0.

WizBangCrash, 2019-11-10:
I have the same problem.
I can't find any alternate sensors to buy that have the 2.5mm phono connector and are waterproof for outside use, but get exactly the same error as @s3phy.
I've tried Ravencore 1.2 and HAA 0.6.0 with same results.

RavenSystem, 2019-11-11:
Duplicated: https://github.com/RavenSystem/esp-homekit-devices/issues/586

closed 2019-11-11 by RavenSystem