RavenSystem/esp-homekit-devices #589: Sonoff Basic R2 ds18b20

ahg675, 2019-11-08:
Hi 
I have sonoff basic r2 with ds18b20 on gpio2 .
How to change gpio14 to gpio2 in this template:

Sonoff Basic/SV/TH10/TH16 HomeKit Switch and Temperature with DS18B20 sensor at GPIO 14:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":22,"g":14,"n":3}]}

RavenSystem, 2019-11-08:
Replace "g":14 with "g":2  ;)

ahg675, 2019-11-08:
One more question
It is possible to make
Sonoff TH16 HomeKit Switch plus Temperature and Humidity sensor at GPIO 14

RavenSystem, 2019-11-08:
Theorically, yes. Never try.

ahg675, 2019-11-08:
"n": Sensor type:
DHT11.
Default. DHT22.
DS18B20.
Si7021 (Not recommended).

RavenSystem, 2019-11-08:
Sorry, I read your question better. Answer is: sure, you can.

ahg675, 2019-11-09:
Working with n2 but only shows temp
Here is my example:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":22,"g":14,"n":2}]}

RavenSystem, 2019-11-09:
DS18B20 is only a temperature sensor, not humidity.

ahg675, 2019-11-09:
I have DHT22

ahg675, 2019-11-09:
Sonoff Basic R2 ds18b20 is working now
I have another sonoff th16 with DHT22

ahg675, 2019-11-09:
Sorry for mix-up

RavenSystem, 2019-11-09:
You must use type 24 instead 22:

https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#type-24-temperature-and-humidity-sensor

ahg675, 2019-11-09:
Thank You very much

RavenSystem, 2019-11-09:
You are welcome. 😉

ahg675, 2019-11-09:
Sorry but I'm still here :)
I did changes but humidity not show at all

TH16 plus DTH22

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":24,"g":14,"n":2}]}

ahg675, 2019-11-09:
Can You correct me, please

RavenSystem, 2019-11-09:
Have you reset Homekit ID of your device?
 
> If you make changes in your JSON that involve accessory types, number of used accessories or change order of them, you must check "Reset HomeKit ID" from "Setup mode" and remove your device from HomeKit; and then you must pair it again.

https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#json-config

ahg675, 2019-11-09:
Thank You again
Now I understand everything

RavenSystem, 2019-11-09:
Is it working fine now?

ahg675, 2019-11-09:
Yes

RavenSystem, 2019-11-09:
Good!

closed 2019-11-09 by RavenSystem