RavenSystem/esp-homekit-devices #590: Sonoff Mini - bistable wall switch - HAA

Fudalii, 2019-11-08:
Hello

I connected a bistable button to sonoff Mini (not momentary/impulse)
Configuration 

{"C": { "l" 13, "b": [{"g" 0, "t" 5}]}, "a": [{"0": {"r": [{ "g": 12}]}, "1": { "r": [{"g" 12, "v" 1}]}, "b": [{"g": 0}, { " g '4}, {"g" 4, "t" 0}]}]}

Works well. 

But if the Wall Button is turned on (**down**) and I turn off the light in the HomeKit application, then the logic reverses. If someone pushes the wall switch up (it should still be OFF), the light turns on ;( 

How to set it to always be OFF when the wall switch is in the "OFF" position?

Bene2103, 2019-11-09:
I think this is the expected behavior. 

You could rewire your switch and Sonoff that way that you cut power from Sonoff and set init state to ON

Fudalii, 2019-11-09:
When you switch the "bi-stable switch" in the **up** position (and in the HK application it was **OFF**) it should still be OFF. 

Identically the other way round. If it is OFF and you switch down (and it was turned on in the application) it should still be turned on.

I hope that you understand me....

Sonoff mini doesn't have 230V on S1 and S2 (3.3V) - But of course everyone must be careful!

RavenSystem, 2019-11-09:
You must use f0 and f1 instead b.

Fudalii, 2019-11-10:
I can not deal with it. I don't quite understand these f0 and f1 ;(

RavenSystem, 2019-11-11:
I must say that is a very rare setup, but here is:

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"f0":[{"g":4}],"f1":[{"g":4,"t":0}],"b":[{"g":0}]}]}`

Fudalii, 2019-11-11:
That's exactly what I meant! Thanks a lot!

closed 2019-11-11 by Fudalii