RavenSystem/esp-homekit-devices #591: Magic Home RGB - Transition very flickery

Bene2103, 2019-11-09:
Some time ago I've tried RavenCore with the MagicHome controller but I wasn't satisfied.
Now I've tried HAA everything works pretty well but the transition from OFF <-> ON or when the Color or Brightness is changed is pretty flickery.

In the video you can see:
RED -> GREEN -> Brightness change -> Brightness change -> RED -> OFF
https://youtu.be/s9Su8YVcwQM

My JSON is the one you provided in the wiki only with changed pins for the colors.

Is there anything I could do about that?

Thank you

RavenSystem, 2019-11-09:
What HAA version are you using?

Bene2103, 2019-11-09:
im using 0.6.2

RavenSystem, 2019-11-09:
Many thanks. I must fix it.

RavenSystem, 2019-11-09:
Try v0.6.3.

Bene2103, 2019-11-09:
You’re lightning fast!

Transitions are super smooth but there’s one more thing. Wenn turning on the brightness overshoots and gets a little bit down later. 

It’s like this:
Turning on brightness Fades to 100%
Brightness 100% for 0.5sek
Brightness goes to 95%

I’ll take a video of this behavior when I’m back home. 

Thank you!

RavenSystem, 2019-11-09:
Can you try v0.6.4?

Bene2103, 2019-11-09:
I tried 0.6.4 transitions are faster now but the brightness still overshoots

https://youtu.be/QbmuD2NQBA4

Thank you!

Bene2103, 2019-11-09:
Found another bug maybe the are related together. 
Setting blue at full brightness -> LED's are switched off an flashing. Red and green working fine.

https://youtu.be/CU5CphsJKV8

RavenSystem, 2019-11-09:
Try 0.6.5.

Bene2103, 2019-11-09:
You nailed it!

Everything is working fine now.

Thank you very much

RavenSystem, 2019-11-09:
Yeah!!!!!! Many, many thanks for your tests and videos!

closed 2019-11-09 by RavenSystem

mafyata, 2019-11-09:
Can you post your json to test on mine magic home rgb led controller?

Bene2103, 2019-11-09:
It’s from the HAA template page in the wiki

{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}]}]}

You need to edit the gpio to your actual pinout

mafyata, 2019-11-09:
I think mine is like your.
Thanks

mafyata, 2019-11-10:
Hmmm, I must change rgb colors in json , but can’t enter setup mode. I toggle 10 times on/off the led strip , but don’t enter in setup mode.

mafyata, 2019-11-10:
How to setup dummy switch for entering setup mode?
I want to include in this json : {"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,”s”:5,"r":5,"g":12,"v":13,"b":[{"g":0}]}]}

Bene2103, 2019-11-10:
I don’t know exactly.

But I do it this way: press the HomeKit button more than 10 times keep pressing until the device gets unresponsive. Usually I press the button about 20 times. When the device is unresponsive you can connect to its access point HAA-XXXX

mafyata, 2019-11-10:
I tried 10 times, not 20.

mafyata, 2019-11-10:
Now i tried 20 times , 25 times and nothing happens. 
This is mine json:
{"c":{"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}

mafyata, 2019-11-10:
I have many devices with ravencore , not haa because they are hardly accessible fisically. If I can’t enter setup mode without fisically access the devices I will not update to haa.