RavenSystem/esp-homekit-devices #592: MagicHome LED Dimmer

mientki, 2019-11-09:
Hello

Is it possible to set HAA as a bulb with a dimmer? In the sample templates I can only see MagicHome RGBW and RGB. What should JSON look like? I have MagicHome LED Dimmer with single channel for monochrome led strip.

RavenSystem, 2019-11-09:
Declare only "w" pin.

closed 2019-11-09 by RavenSystem

bj-tracer, 2019-11-11:
Hello!
How to reset that bulb with HAA (or going to setup mode)?
This device hasn't buttons. You need to switch on/off it 10 times for reset with original firmware.

mientki, 2020-01-01:
I have MagicHome single channel and I'm trying to set up HAA. I don't know what I'm doing wrong, but configuration doesn't work. Please help. My configuration:

```
 {
   "c": {
     "b": [{
       "g": 0,
       "t": 5
     }]
   },
   "a": [{
     "t": 30,
     "w": 12,
     "b": [{
       "g": 0
     }]
   }]
 }
```

Correct settings of this device on other software:
(https://github.com/curla92/LED-Dimmer-Homekit)

[image]