RavenSystem/esp-homekit-devices #593: Sonoff 5V - A Fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

jakudo, 2019-11-09:
Hi,

I'm a newbie here. I'm trying to flash Sonoff 5V - to work with HomeKit for my Gate.
Here is how I connected it:
[image]

I've tried to put it into the Flash mode as described here:
https://www.youtube.com/watch?v=E0WBoFCcnYY

and it does the same as in the video.

However, when I try to run command:

 `esptool.py -p /dev/ttyUSB0 erase_flash`
it starts connecting, shows several .......______.......... and then 
`A Fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`

Any idea what am I doing wrong?

Bene2103, 2019-11-09:
Are you using the right port?

try "ls /dev/" and see what the right port is. It should look something like "tty.usbserialxxxxx"

What OS are you running?

jakudo, 2019-11-09:
sorry, forgot to mention it. I use Ubuntu 19.04

The only difference between "ls /dev/" before and after I plug it in is "ttyUSB0"

Bene2103, 2019-11-09:
Ok then this device should be the right one.

Maybe RX and TX are switched?

jakudo, 2019-11-09:
Tried that too with the same result.

jakudo, 2019-11-14:
Hello,

so for other I found what the probelem was. The pins 1 and 4 on the first picture was ok, however 2 nad 3 was incorret.
The Tx and Rx should be connected directly on the chip (see red rectangle) on 5th and 6th pin from button.
So it might help somebody.
[image]

closed 2019-11-14 by jakudo