RavenSystem/esp-homekit-devices #594: esp8266 with button + motion + temp/hum

vsidlu, 2019-11-11:
I have this json - 
{"c":{"o":1,"l":2,"i":1},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0,"p":1,"t":1}]},{"t":12,"i":15,"f1":[{"g":13}]},{"t":24,"g":14}]}

without {"t":24,"g":14} everything is working perfectly. but when I add a:t:24, motion sensor is not triggered immediately. where is problem?

RavenSystem, 2019-11-14:
Have you try last version?

vsidlu, 2019-11-15:
i think found solution - need to add "p" to motion json - 
{"t":12,"i":15,"f1":[{"g":13,**"p":1**}]}

found in this post - https://github.com/RavenSystem/esp-homekit-devices/issues/549#issuecomment-545025197

vsidlu, 2019-11-27:
hi,
i tested two identical sonoff basics with motion sensors attached-
first with HAA - v0.2.3
second with HAA - v0.7.4 (I think starting from v0.3.2 there is changes in code)

First one works without problems and respond almost immediately.
Second one responds with delay if responds at all. Tried out different json options but nothing helps (pullup deactivated, inverted position). Log do not shows any problem. I see motion is activated but with unreasonable delay.

vsidlu, 2019-12-11:
hi,
right now testing this json - {"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"p":0,"i":10,"f1":[{"g":12,"i":1}]},{"t":24,"g":14}]}

decalerd "p" outside "f1" is working. but motion is triggering when "g":12 returns high.

RavenSystem, 2019-12-11:
p outside f1 do nothing, remove it. And in order to your sensor works, remove i into f1 because your sensor is not inverted.

vsidlu, 2019-12-11:
still do not have explenation but this json - {"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"i":10,"f1":[{"g":12}]},{"t":24,"g":14}]} is not wokring!

vsidlu, 2020-01-02:
{"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"i":5,"f1":[{"g":12,"i":1}]}]}
video - https://youtu.be/gb9iH0NSHWQ
{"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"i":5,"f1":[{"g":12,"i":0}]}]}
vidoe - https://youtu.be/C7srbeZxcZs

when i=1 motion is triggered every time
when i=0 motion is not triggered at every movement after some time passes

vsidlu, 2020-01-10:
{"c":{"o":1,"l":2,"i":1,"b":[{"g":0,"p":1,"t":1}]},"a":[{"t":12,"i":5,"f1":[{"g":12,"p":0,"i":1,"t":0}]}]}

i do not know what "t":0 do but it seems everything is working correctly.

@RavenSystem commented in this threat https://github.com/RavenSystem/esp-homekit-devices/issues/632#issuecomment-573026272

closed 2020-01-10 by vsidlu