RavenSystem/esp-homekit-devices #595: JSON for Shelly1 switch with external stateless button

elcatedratico, 2019-11-12:
Hello,

I'm having trouble to setup the right JSON string for my shelly1. The template for a switch with external button works great. Now I want to use a external stateless button (ok, it is more than one because of my changing circuit). Can someone provide me an example?

Hornett69, 2019-11-12:
Hello,
this post can help you...
[https://github.com/RavenSystem/esp-homekit-devices/issues/411](url)

Hornett69, 2019-11-12:
Hello,
This post can help you...
[https://github.com/RavenSystem/esp-homekit-devices/issues/411](url)

Hornett69, 2019-11-12:
hello,
to make a frame...
[https://glumb.github.io/haa-configurator/](url)

and this post can help you...
[https://github.com/RavenSystem/esp-homekit-devices/issues/411](url)

elcatedratico, 2019-11-13:
your links doenst work

Hornett69, 2019-11-13:
Sorry for that... i don’t know why but...
1# Don’tuse the link... Copy / Paste this link in you browser to access for HAA configurator
https://glumb.github.io/haa-configurator/

2# And for the second link, search in github : JSON config 411
Look this issue is closed but it’s an exemple for Shelly 1 Stateless... May be this one can help you.

elcatedratico, 2019-11-13:
All right... I got it working. BUT, how to configure the shelly? In Home App and EVE App I have to assign a scene or another device. I want to switch the physical output of the shelly.
If I try to set up a scene, I can't select the shelly as a actuator. 

Has anyone an idea?

Hornett69, 2019-11-13:
I know the problem because I have the same with a sonoff Basic.  I solved it using a normal switch with a one-second timer to return to the initial state.  I was able to configure the physical output of Sonoff.

elcatedratico, 2019-11-13:
Isnt there a way to create a JSON string that does exactly that? I'm currently playing with the configurator ... but I have no idea what I'm doing ;)

elcatedratico, 2019-11-14:
> I know the problem because I have the same with a sonoff Basic. I solved it using a normal switch with a one-second timer to return to the initial state. I was able to configure the physical output of Sonoff.

Can you send me your JSON?

elcatedratico, 2019-11-14:
When I look at all the JSON for stateless button, there is no output gpoi definded. 
I struggle defining the same input (value) to two different output values. Maybe José knows how to do that

Hornett69, 2019-11-14:
Yes José is the best to answer at this project...

My JSON frame for my Sonoff Basic :
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0,"i":1},{"t":5,"f0":[],"f1":[{"g":2,"t":1,"p":1,"i":1}],"s":0,"b":[],"i":5}]}

Switch to activate relay on GPIO12 with timer 5s
With contact sensor closed on GPIO2 with timer 5s to reset in initial state

elcatedratico, 2019-11-14:
@RavenSystem I know you are probably very busy, but do you have any idea? 

> When I look at all the JSON for stateless button, there is no output gpoi definded.
> I struggle defining the same input (value) to two different output values. Maybe José knows how to do that

RavenSystem, 2019-11-14:
Question here is: What thing is a "stateless button"? Because there is a Homekit accessory type called stateless button, or events button. But this accessory type has not any physical output. Other thing is a normal button instead a switch.

elcatedratico, 2019-11-15:
That's the right question. I mixed "stateless button" and "momentary switch" (normal button). Sometimes Translation is a bitch ;)
I want to use the shelly with a latching switch (normal button instead a toggle switch) in multiway switching circiut.

elcatedratico, 2019-11-15:
Thank you very much.... works like charme

elcatedratico, 2019-11-20:
Sorry to revive the thread again. BUT i noticed a wierd problem with the current configuration. Sometime I need to push the (momentary) button twice to activate or deactivate the light. This happens after the button wasn't actuated for a while. As far as I figured out, it takes exactly 10s to change from one-click to doubble-click, independed if the light was triggerd by the app or not.
@RavenSystem is this a known issue or a new bug? I'm running the latest software (0.6.18) but is also occured at 0.6.13

RavenSystem, 2019-11-20:
It seems that your button is inverted. Try this:

`{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0,"i":1}]}]}`

closed 2019-11-28 by RavenSystem