RavenSystem/esp-homekit-devices #596: Sonoff Basic R2 with Contact Sensor

Hornett69, 2019-11-12:
Hello and congratulations for your excellent work that interests me a lot for my IoT projects. My first project is to command the Gate from with a Sonoff Basic R2. I made a switch and I drive my Gate with a pulse of one second. It works perfectly.
I added a Contact Sensor on GPIO2. I would like homekit to show if its open or closed state.
The only configuration I was able to do was to detect Contact Sensor closed with a 10 second delay to return to the initial state.

Configuration:
- Sonoff Basic R2 with an ESP 8285 chip
- Contact sensor
- HAA installed in version 0.6.6

JSON frame:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0,"i":1},{"t":5,"f0":[],"f1":[{"g":2,"t":1,"p":1,"i":1}],"s":0,"b":[],"i":5}]}

I would like to use the following JSON frame:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0,"i":1},{"t":5,"f0":[{"g":2,"t":1,"p":0,"i":0}],"f1":[{"g":2,"t":1,"p":1,"i":1}],"s":0,"b":[]}]}

I may not have understood how it works ... Could you help me a little ?
Thanks.

RavenSystem, 2019-11-14:
`"f0":[{"g":2,"t":1,"p":0,"i":0}],"f1":[{"g":2,"t":1,"p":1,"i":1}]` Here, "p" is for pullup-resistor and "i" is for inverted. Only the first declaration of both will take effect. Then, you use same type "t":1 and it is wrong. One must be 1 but other must be 0; they are the 2 possibles sensor positions (contacted and no contacted). 

Try this (I removed default values)
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"i":1},{"t":5,"f0":[{"g":2,"t":0}],"f1":[{"g":2}]}]}`

Other thing: using gpio 0 or 2 (this case) for a sensor is not a good idea.

Hornett69, 2019-11-15:
The JSON frame works very well ... a big thank you for your help because my problem is solved thanks to you.  Why is the use of GPIO0 or 2 not a good idea?  It works very well ...

RavenSystem, 2019-11-15:
GPIO 0 and 2 must be in high state when device boots, because if not, device will enter in flash mode.

Hornett69, 2019-11-15:
OK, I did not know ... What is the alternative ?

RavenSystem, 2019-11-15:
You can use GPIO 3.

Hornett69, 2019-11-15:
ok, thanks for the recommendation

closed 2019-11-15 by Hornett69

cristidobrescu, 2022-03-05:
hi @Hornett69 
would like to tell me where did you put the magnetic contact on the board? a pic would help

Hornett69, 2022-03-06:
Now I use Sonoff SV with GPIO3… Not rally possible on R2

cristidobrescu, 2022-03-06:
ok, thanks