RavenSystem/esp-homekit-devices #597: New free windows flasher

vglucas, 2019-11-12:
HomekitFlasher is a simple application for Windows desktop environments developed for free and downloadable from https://github.com/vglucas/HomeKitFlasher. An excellent tool dedicated to those who are not familiar with the terminal and want to make, and speed up, the flash of their device. The main and predefined actions of the system also allow the Ravencore firmware flash, developed by José A. Jiménez and Life Cycle Manager called OTA (acronym of Over The Air) and created by HacK (HomeACcessoryKid). As we will see below, the program also allows third-party firmware flashes thanks to a simplified interface.

For any signaling from the menu? - Report and send your report

than-sad, 2019-11-13:
Hi, I bought 8 sonoff basic r3 and searching on internet I found one website using this app like the only one who's able to flash this devices with old Ravencore. I bought a USB adapter based on FT232RL from FTDI, got an old laptop, installed windows, installed this app, installed the USB adapter drivers, and when I Tried to flash the devices, the app didn't recognized the FTDI adapter. Finally I went back to my mac and done it easier than expected just with two command lines (delete, and flash). Supposedly the only way to flash this devices was flashing first the firmware with blank_config.bin in three different addresses after erase it. Probably was my fault because I didn't understand it properly (wasn't my language (es) or english). I'm absolutely newbie in the flashing firmwares world.

It's just my experience, I'm sure this app can be really useful for a lot of people. Wasn't my case.

Best regards,
Than.

vglucas, 2019-11-13:
Hi, there are some devices that are not detected by windows, the manufacturer of the FT232 has included in its latest Driver a control if the FT232 is not a fake one. If it finds it falsified it resets the VID with which Windows recognizes the USB device. perform this procedure or from the menu? and SEGNALA, send a message to me and I will be happy to help you
[image]

than-sad, 2019-11-14:
Hi, I downloaded signed drivers from website, connected the device and after detected by windows installed it and checked in device manager if the device was present and with no conflicts. In your app, tried several times to detect it but always with message 0 devices found. Anyway is windows, there are thousand reasons to fail. Thanks for your interest in help. Im a happy Mac user (hackintosh in my case) and to be honest every time I run Windows OS I finish frustrated and tired of install eternal updates and stupid errors.

Best regards.

batista16, 2019-11-19:
I also have issues with this, no devices found.

vglucas, 2019-11-19:
what kind of device do you use? perform the procedure described above
[image]

batista16, 2019-11-19:
[image]

This is the error that I get when I try to scan. 

I have a FTDI232, and the device works with other methods.

vglucas, 2019-11-19:
did you run the full scan as described above ? Open a report from the menu ? and then SEGNALA it and I'll contact you and I'll be happy to help you

vglucas, 2019-11-19:
I made changes, can you update and check?

diodo157, 2019-11-19:
Hi everyone ... tried the program is really very nice ... easy and intuitive ... it works fine for me. for those who have some problems, I suggest you uninstall Pyton and also delete the files that remain in the folder, then reinstall everything. Really very happy to have found this program

diodo157, 2019-11-19:
[image]

batista16, 2019-11-20:
now I see this: 
[image]

I see an FTDI there but doesn't look like the COM port.

vglucas, 2019-11-20:
I have found that the FTD2XX chips are not detected by the program, I am working fully on this.

maxnasva, 2019-11-21:
hello,I tested the program successfully with these devices:
CP2102 chip usb interface
nodemcu (3 item different)
arduino
see attached photos
[image]
[image]
[image]

slidset, 2019-11-23:
I used the program with the serial converter FTD1232, it works.
just download the drivers:
https://www.ftdichip.com/Drivers/CDM/CDM21228_Setup.zip
[image]

closed 2019-12-19 by vglucas