RavenSystem/esp-homekit-devices #598: Device freezed after 0.6.10 update

macjl, 2019-11-12:
Hello,

I have updated some of my sonoff devices today. I have updated them to the version 0.6.10 before seeing there were an issue with it :

> Fixed a bug introduced in v0.6.9 that resets device when a sensor is activated.

I was able to update most of the devices to 0.6.11, but one of them seems to be freezed now. It is not accessible from Home App, doesn't respond ton ping, and I'm not seeing any Wifi SSID from it. I tried to switch off and on the power without luck.

Should I have to re-flash to recover it? Or is there another way? The sonoff is not very accessible, and if there is an other solution it would be great.

mafyata, 2019-11-12:
Did you try to power off and on again?

mafyata, 2019-11-12:
I had one frozen device and when I unpluged and pluged again to the wall ac source it work again.

macjl, 2019-11-12:
I tried it multiple time without success.

mafyata, 2019-11-12:
Check in home app to add accessories. I had device waiting to be added after power failure with haa 0.5

macjl, 2019-11-12:
Thank you for your help, but still no success with this...
I think I will have to re-flash it.

than-sad, 2019-11-13:
Hi, I had similar problem with one TH16 in Thermostat mode. After OTA update, it didn't responded until I checked trying to add new device in Home and detected it as a new one. Deleted the old one and problem solved.

Regards, Than.

RavenSystem, 2019-11-14:
Sorry for the bug update. But it only affected to accessories with sensors (but not temperature sensors). In my case, I cut general power and reposer it, and then I updated my faulty device using OTA.

macjl, 2019-11-15:
It was a Sonoff Basic with switch and stateless button on GPIO 0 + contact sensor on GPIO 2.
I have re-flashed it and it’s working now.

closed 2019-11-15 by macjl

RavenSystem, 2019-11-15:
Using a contact sensor in gpio 2 is not a good idea, because gpio 2 must be high when device boots.

macjl, 2019-11-16:
I know that. But this contact sensor is normally high, except when someone push the doorbell. So the reboot go right most of time.
Do you know if I can use Rx or Tx pin instead of GPIO2 ? What are their PIN number ?

RavenSystem, 2019-11-17:
@macjl Ok, I only told you it ;). You can use GPIO 3 if you want (I think it is RX).

macjl, 2019-11-18:
Thank you!
For my future devices, I will prefer the use of GPIO 3.