RavenSystem/esp-homekit-devices #600: Device doesn't come online

tonysprenk, 2019-11-13:
Hi, 

First of all thanks allot for everyones efforts in this project! Really cool...if I can get it to work (its probably me, not you).

So I'm trying to get my Tuya wall sockets (built into the wall) to work natively with Homekit.

Heres what I've done so far:

-Used [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert) to flash the ESP8266 chip with Espurna.
-Used ArduinoIDE to upload the BasicOTA file.
-Used espota.py to upload blank_config.bin, rboot.bin and Otaboot.bin from this repository.

After uploading blank_config.bin, rboot.bin and Otaboot.bin I waited a while for the LCM-XXXX SSID to come online but it never came online, I power cycled the wall socket but nothing. 

Could anyone help me troubleshooting to figure out what I did wrong? 

I'm trying to do all of this OTA because the sockets are built into my wall and I have a lot of them and I would rather not have to disassemble them if I don't have to. 

Thanks a lot!

mafyata, 2019-11-13:
Did you erase the chip before upload blank_config.bin, rboot.bin and Otaboot.bin ?
esptool.py --port com4 --baud 115200 erase_flash
com4 is for example, your’s may be 3 , 5 .....

tonysprenk, 2019-11-13:
Thanks for the quick reply! 

I’m using espota.py not esptool.py, so I don’t think I can erase the chip as it needs the BasicOTA file on there to connect to my WiFi to be able to us espota.ty. 

If the chip being completely blank is a prerequisite I guess it’s not possible to flash the blank_config.bin, rboot.bin and Otaboot.bin file OTA?

mafyata, 2019-11-13:
I don’t kwon what are you using. Is it wired connected to pc?

mafyata, 2019-11-13:
I read about espota.py
I think you can’t flash to ravensystem with espota.py

tonysprenk, 2019-11-13:
No I’m trying to do the whole process OTA(over the air), wireless, over my WiFi network.

mafyata, 2019-11-13:
I think that this is inpossible. Tuya is using wifi chip to connect to wifi to be able to upload firmware , but how to broadcast wifi to connect to ravensystem(LCM-XXXX SSID).

tonysprenk, 2019-11-13:
I am able to flash OTA Tasmota or espurna with Tuya concert so...

RavenSystem, 2019-11-14:
My firmware uses LCM, and I think that it is not compatible with Tuya system.

closed 2019-11-14 by tonysprenk