RavenSystem/esp-homekit-devices #603: same accessory password, setupID and category

mgoeller, 2019-11-14:
Hi Jose,
I see that you are using the same accessory password to pair, the same setupID and the same category code (in case one configures twice a Lightbulb for example). How did you resolve this?

If I am trying to do so, my pairing request gets send to the wrong accessory and it basically tries to pair with an already paired one. If I temporarily power off the paired one, afterwards it has "no response" in the Home App.

Thank you!

RavenSystem, 2019-11-14:
HomeKit should detect unpaired devices. I have not any issue with that. When you try to paired one, tell Home App that you have not any code or you can not scan it.

mgoeller, 2019-11-15:
It was my fault, I removed the random additions in the end to the advertised mDNS name... Thank you!

closed 2019-11-15 by mgoeller