RavenSystem/esp-homekit-devices #604: Problem V.6.14 on magic home led rgb

mafyata, 2019-11-14:
Hi , I updated to v6.14 and my led strips have a problem. The light is very very low. I don’t have a problem when I was on v6.11
[image]

mafyata, 2019-11-14:
This is my config and was working good.
 {"c":{"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}
Can I downgrade to v.6.11 ?

bj-tracer, 2019-11-14:
https://github.com/RavenSystem/haa/releases/tag/0.6.11 - You may download main.bin and install it without OTA.

mafyata, 2019-11-14:
Thanks. I prefer ota . Where may be is the problem? I tried “p”,”d” actions , but no success, same result.

bj-tracer, 2019-11-14:
p & d  is for transitions effects
try to use RGBW Factor

mafyata, 2019-11-14:
Tried just now , nothing happens.

mafyata, 2019-11-14:
And i am seeing a mistake. Auto dimmer has 2 option with same initial - “d”

kutdenis, 2019-11-14:
Same problem, but on H801. Very faint light. 
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

mafyata, 2019-11-14:
With 6.15 is good , but when changing brightness or on/off it is blinking brighter. On 6.11 was perfect.

RavenSystem, 2019-11-14:
@mafyata Can you try this to know if issue is PWM freq?

`{"c":{"q":65535,"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}`

mafyata, 2019-11-15:
I will try it later and will write the result.

mafyata, 2019-11-15:
No, no changes .

RavenSystem, 2019-11-15:
@mafyata And with this?

`{"c":{"q":1000,"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}`

mafyata, 2019-11-15:
No

RavenSystem, 2019-11-15:
@mafyata Try last version 0.6.16 with your original JSON:

`{"c":{"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}`

mafyata, 2019-11-15:
Yes , now is ok

mafyata, 2019-11-15:
On 0.6.15 it was blinking brighter when I was set it to 100%. Is it mean that my led strip was not lighting at 100%?

RavenSystem, 2019-11-15:
@mafyata it depends of selected color. To keep brightness stable, when you select white, color channels are at 33% and white is at 100%. You can play with factor options, depending what you want:

For example, try this:
`{"c":{"b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13,"fr":3,"fg":3,"fv":3}]}`

closed 2019-11-18 by RavenSystem