RavenSystem/esp-homekit-devices #607: Template Help

tonysprenk, 2019-11-14:
Hi,

Thanks everyone so much for the hard work on this! I have managed to flash this firmware on some Tuya Sockets using [Tuya-Convert](https://github.com/ct-Open-Source/tuya-convert) but now I need some help with a template for a Tuya smart socket. 

Not sure if it will help but [this](https://blakadder.github.io/templates/moeshouse_smart_socket.html) is the Tasmota template for the same socket. 

Hopefully someone can point me in the right direction!

Thanks again!

tonysprenk, 2019-11-14:
For anyone that's interested I can do a small write up of how I got esp-homekit-devices flashed OTA (!!!) onto a Tuya device.

RavenSystem, 2019-11-15:
This is amazing! Yes, please, write it.

Have you tried this template?
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#tuya-swa1-with-esp8266-smart-plug

tonysprenk, 2019-11-15:
https://www.reddit.com/r/HomeKit/comments/dwiaws/flashing_tuya_devices_to_be_supported_natively/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

tonysprenk, 2019-11-15:
For those curious: I changed the flash mode of rboot.bin and combined it with the other 2 binaries offered by this repository. I was then able to flash the Tuya device using Tuya-convert. Pretty simple in the end!

tonysprenk, 2019-11-15:
@RavenSystem the template works in all aspects accept for it doesn’t turn the relay on. The physical button changes the color of the led and the state in home app and vis verse the same with the button in the home app but the relay isn’t switched on or off. What should I change?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RavenSystem, 2019-11-15:
GPIO 5 is not the relay. It seems to be only status LED. We must find GPIO relay. Maybe an Internet search helps.

gerardwr, 2019-11-15:
@tonysprenk Nice !

Could you please share the details on how to combine the 3 files into one?

RavenSystem, 2019-11-15:
@tonysprenk Please, can you share tuya-firmware version using otaboot.bin? I could include it in the repo.

RavenSystem, 2019-11-15:
@tonysprenk What is your device? Here there is GPIO info about Tuya devices: https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)

kueblc, 2019-11-16:
Hi @RavenSystem, would you like your firmware bundled with tuya-convert? I'm sure there are others who would appreciate your HomeKit integration, thanks @tonysprenk for the tip off

tonysprenk, 2019-11-16:
@RavenSystem @kueblc 

its the moeshouse I think relay should be GPIO12. 

We should look into building some safeties into the HK firmware because it’s designed to be flashed by wire of anything goes wrong the device is bricked and there is no return. Like the wrong WiFi password. Or I for example uploaded a wrong template and now one of the devices is bricked as I can’t put it in setup mode anymore. @RavenSystem  is this correct?

peros550, 2019-11-16:
@tonysprenk 

I'm very interested in flashing HAA on a Blitzwolf BW-LT11 (a very cheap but nice color led strip RGBW). 

Which firmware did you use to flash? I saw on reddit page you gave a link for a thirdparty firmware bin file, is this HAA? 
After flashing do you have OTA functionality for getting HAA/Ravencore updates?
After flashing, are you able to use same method to get back to original firmware?

Thanks & Regards

tonysprenk, 2019-11-16:
Yes the binary I linked is a modified version of HAA. It includes the 3 binaries offered in this repository. After flashing every thing is done OTA. There is no return to original firmware after flashing. Make sure you know the different GPIO pins for R, B, G and W and any other buttons or anything your device might have so you can make a working HAA template. Let us know your results!

mafyata, 2019-11-16:
This is the command to backup your original firmware:
esptool.py -p com4 -b 115200 read_flash 0 0x200000 c:/firmware.bin

peros550, 2019-11-16:
> This is the command to backup your original firmware:
> esptool.py -p com4 -b 115200 read_flash 0 0x200000 c:/firmware.bin

thanks! but I wouldn't be able to use same method to go back right? I would probably need to solder wires directly to the chip.

mafyata, 2019-11-16:
I don’t know which method will you use and need you solder the cables. Sonoff basic,th no need to solder , magic home led controller need to solder. I don’t know your chip.

peros550, 2019-11-16:
Soldering or not, it looks that going back via OTA or TuyaConnect is not possible once you flash a different firmware. But thanks for the command to save the original firmware. It will be handy.

tonysprenk, 2019-11-16:
Tuya-Convert also backs up the original firmware and gives you a chance to put it back (undo) before you do the final flash.

RavenSystem, 2019-11-17:
@kueblc Yes, I think installing HomeKit firmware using Tuya-Convert is a good idea for those compatible devices. But I think the best choice is using otaboot.bin to allow then to update it through OTA.
https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#installation-with-ota

RavenSystem, 2019-11-17:
@tonysprenk Try this:

`{"c":{"b":[{"g":13,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":5},{"g":12}]},"1":{"r":[{"g":5,"v":1},{"g":12,"v":1}]},"b":[{"g":13}]}]}`

tonysprenk, 2019-11-17:
@RavenSystem thanks! I will try this out. Like I said in my previous comment I have one plug that doesn’t work anymore. The led is red and flashes once every 30 seconds. I think in the previous template I tried the button sequence to put it into setup mode was 5: hold indefinitely but no matter how long I hold the button nothing happens. I also have one plug that is in setup mode, when I connect to the AP and put in al the credentials it seems to never install the HAA firmware and just keeps putting out the AP. Any ideas?

RavenSystem, 2019-11-17:
@tonysprenk Did you installed otaboot.bin or directly HAA main.bin?

tonysprenk, 2019-11-17:
otaboot.bin as part f the tuya_convert flash and main.bin once the AP is created.

closed 2019-11-29 by RavenSystem