RavenSystem/esp-homekit-devices #608: Support for Window Covering

tonysprenk, 2019-11-15:
Hi, I have a few [curtain devices] (https://s.click.aliexpress.com/e/n3g3Q1For) that I would like to flash to be compatible with HomeKit. Could anybody point me in the right direction?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RavenSystem, 2019-11-15:
RavenCore firmware with Shelly 2.5 devices is compatible, but you will need HAA firmware to do what you want. Problem is that HAA is not compatible with coverings yet. You must wait.

tonysprenk, 2019-11-15:
Ok! Let me know if I can do any testing! 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RavenSystem, 2019-11-15:
Sure! But first of all, it is necessary to know the GPIOs used by those devices.

mientki, 2019-11-17:
is there a chance of support for window blinds and shelly 2.5 in the near future? RavenCore is very difficult for beginners.

mafyata, 2019-11-17:
Haa is more difficult.

tonysprenk, 2019-11-18:
https://github.com/arendst/Tasmota/wiki/Zemismart-Curtain-Motor this is the device. hopefully it helps.

mientki, 2019-11-18:
> Haa is more difficult.

it is not more difficult if I configure the device :) and a friend will only add it to their home application.

mafyata, 2019-11-18:
Yes , but more people here will flash, configure and you use it without friend.

RavenSystem, 2019-11-18:
If a JSON template exists, HAA is easier than RC. But that is not the matter, because RC is outdated, deprecated and is no longer developed. Only Garage Doors and Window Covers left to migrate to HAA. And when you control your air conditioner or your TV using only an ESP-01 with an IR-LED, you will forget RC very quickly.

closed 2019-11-29 by RavenSystem

mientki, 2019-12-01:
We are all patiently waiting for support for windows blinds. :)