RavenSystem/esp-homekit-devices #609: One click activation of thermostat devices

macjl, 2019-11-15:
Hello,

I think it would be great if we can turne on/off thermostat devices with just one click in Home App. It has been done in the homebridge-daikin-local plug-in, and working very well for me : 
(https://github.com/cbrandlehner/homebridge-daikin-local/issues/15)

Do you think it's possible to implement it?

Regards

RavenSystem, 2019-11-17:
I think you can do it declaring an additional switch and using automations.

macjl, 2019-11-18:
It should work, but it is not very user friendly.

The way it has been done in homebridge-daikin plugin was adding this characteristic : 
```
    this.heaterCoolerService
      .getCharacteristic(Characteristic.Active)
      .on('get', this.getActive.bind(this))
      .on('set', this.setActive.bind(this)); 
```
With the 'setActive' action, the thermostat is powered up / powered down with just a click on the icon in home app. Is this characteristic implemented in HAA?

RavenSystem, 2019-11-18:
Ahh, I see it now. I use HomeKit Thermostat Accessory and you are talking about Cooler/Heater Accessory. They are different accessory types. Thermostat is the old, and Cooler/heater is the new. "Active" characteristic is available only at the new version. I want to migrate old to new, but I have not had enough time yet.

macjl, 2019-11-22:
Hello,

I have seen that you implemented it in version 0.7. I'ts working very well for me!
And I was able to remove the fake switch I added to enter in setup mode.

Thank You!!

closed 2019-11-28 by RavenSystem