RavenSystem/esp-homekit-devices #612: MagicHome RGBW - Reset to default (wifi & HomeKit ID)

mientki, 2019-11-16:
Hi
I have a MagicHome RGBW device. I uploaded HAA software and configured JSON. I would like to transfer them to another WiFi network. How do I restore HAA's default settings, reset HomeKit ID and network settings. The device has no physical button and I cannot enter it in setup mode. What happens if the device does not find a known network? Will it go into AP mode? This is not the case for me. I have the same problem with MagicHome single channel. Maybe it is worth introducing the "Reset HomeKit ID & WiFi config" option in the next update? What do you think about it?

mafyata, 2019-11-16:
To reset , you must flash it again. To change id or wifi network you must enter in setup mode. You can enter setup mode with 10 times on and off the led strip from homeapp button.

Bene2103, 2019-11-16:
Theres no need to flash it again, if you are in setup mode you can reset HomeKit ID an set up a new network connection

closed 2019-11-29 by RavenSystem