RavenSystem/esp-homekit-devices #613: Device joins wifi but is not available in home anymore

Hanfrey, 2019-11-16:
Hi all, 

so I flashed a LSC Smart Bulb E14 with Tuya-convert over wifi. 
Now the Bulb is in a "strange" mode. But first I try to explain what I did:

I configured the Lightbulb with this template: 
_Bawoo Lightbulb:_
```
{"c":{"b":[{"g":0}]},"a":[{"t":30,"r":14,"g":12,"v":13,"w":4,"fr":4,"fg":4,"fv":4}]}
```
After that the bulb joined my wifi network. I opened Home added the bulb and wanted to activate it with home, but nothing happened. So I turned off and on the bulb again and she was not reachable anymore from home. So i tried now to get the bulb again in setup mode, which is not working :(. She joins still my wifi. 

What I tried to get in the setup mode again:
- Turn off
- Turn on
- Wait 10sec
- Turn on/off 10 times, so that the last state is on 

Also tried it with 8 Times on/off and then again on.

What is the correct way? Any other ideas?

Thanks!

KR
Mark

mafyata, 2019-11-16:
Must be 8 times if you are with latest firmware

Hanfrey, 2019-11-16:
> Must be 8 times if you are with latest firmware
tried that. But maybe im doing it wrong?

I start in "On".
- Off  (1x)
- On 
- Off (2x)
- On
- Off (3x)
- On
- Off (4x)
- On 
- Off (5x)
- On 
- Off (6x)
- On 
- Off (7x)
- On 
- Off (8x)
Turn on again.

RavenSystem, 2019-11-17:
Each transition from off to on or from on to off counts. And don't worry if last is off. A thing that helps is setting brightness to 1% or 2%, so device needs less time to turn on and off.

RavenSystem, 2019-11-17:
And a LSC Smart Bulb E14 is not a Bawoo Lightbulb.

Hanfrey, 2019-11-18:
> And a LSC Smart Bulb E14 is not a Bawoo Lightbulb.

I knew that, but in Tasmota its pretty easy to change...so I thought, try it out 

> A thing that helps is setting brightness to 1% or 2%, so device needs less time to turn on and off.

So I need to wait between the on and off switches? I just have a Lamp where I can cut physically the power, so it needs to be on or the bulb has no power. I tried now, but still no luck: 

Off
On (1x)
2sec wait
Off
On (2x)
2sec wait
Off
On (3x)
2sec wait
Off
On (4x)
2sec wait
Off
On (5x)
2sec wait
Off
On (6x)
2sec wait
Off
On (7x)
2sec wait
Off
On (8x)

RavenSystem, 2019-11-18:
Of course, lamp must be powered to access Wifi and HomeKit in the way you can turn on and off it.

Hanfrey, 2019-11-18:
> Of course, lamp must be powered to access Wifi and HomeKit in the way you can turn on and off it.

I need to turn off and on at the button of the lamp, because its not available in HomeKit. Is this not possible?

RavenSystem, 2019-11-18:
Device must be always powered on. With JSOn that you used, you must turn on and of from HomeKit App. The other way is to connect GPIO-0 and ground. With last version, an emergency way to enter setup mode is available, but you are not in the last version.

Hanfrey, 2019-11-20:
Is there anyway I get it back in Home? If I scan HAA QR-Code it says its already added and I need to reset the device.

pilot1981, 2019-11-20:
Where I can find QR CODE?

Hanfrey, 2019-11-20:
> Where I can find QR CODE?

https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#homekit-setup

closed 2019-11-29 by RavenSystem