RavenSystem/esp-homekit-devices #614: Can not enter setup mode HAA v0.6.17

vsevolod-volkov, 2019-11-17:
Hi,

Shelly 1 can not enter setup mode with eight-time fast press after HAA update from v.0.5.1 to v.0.6.17. Before update worked well. Neither Hardware button not HomeKit button can not enter setup mode.

How do I enter setup mode? After how many seconds **from power on** should I off device to activate emergency setup mode?

vsevolod-volkov, 2019-11-17:
Comment below may be helpful, I suppose.
On other Shelly 1 with UART-attached terminal I see in log fatal exception diagnostics then rebooting when I attempt to enter setup mode:
```
...
HAA > Checking setup mode call
Fatal exception (28): 
epc1=0x40204995
epc2=0x00000000
epc3=0x4020366d
excvaddr=0x0000002c
depc=0x00000000
excsave1=0x40204a78
Registers:
a0 40204a78 a1 3fff3a20 a2  00000003 a3  3fff1a98
a4  00000001 a5  3fff3a2c a6  00000018 a7  00000000
a8  00000003 a9  60000000 a10 00000000 a11 0000000a
a12 3fff15c0 a13 3fff15b8 SAR 00000020

Stack: SP=0x3fff3a20
0x3fff3a20: 00000003 00000000 00000018 00000000
0x3fff3a30: 00000000 00000000 00000000 00000000
0x3fff3a40: 00000000 00000000 00000000 00000000
0x3fff3a50: 00000000 00000000 00000000 00000000
0x3fff3a60: 00000000 00000000 00000000 a5a5a5a5
0x3fff3a70: 0000014c fffffffc 3fff39d0 00002916
0x3fff3a80: 3fff14b8 3fff14b8 3fff3a78 3fff14b0
0x3fff3a90: 0000000d 3ffefd6c 3ffefd6c 3fff3a78

Free Heap: 37984
_heap_start 0x3fff1e58 brk 0x3fff7b6c supervisor sp 0x40000000 sp-brk 33940 bytes
arena (total_size) 23828 fordblks (free_size) 4044 uordblocks (used_size) 19784
 ...
```

vsevolod-volkov, 2019-11-17:
Entering setup mode in v0.6.16 works well. So that issue was introduced in 0.6.17 as I see.

mafyata, 2019-11-17:
I confirm that. I can’t enter setup mode from home app.

https://drive.google.com/file/d/1_Wud5Dpd4IR4cf4y9IotKuAwIpUDgNoB/view?usp=sharing

mafyata, 2019-11-17:
I see something strange. It appears on about latest 2-3 firmwares. Maybe I can call it “slow responce”. See video

https://drive.google.com/file/d/1Sq-reXIWApDg08G72vWGrg3UWJBuqN-U/view?usp=sharing

RavenSystem, 2019-11-17:
It is a bug. Now it is fixed:

https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_0.6.18

mafyata, 2019-11-17:
Can you explain what is this new function in 6.17:
Added Actions to Events Button Accessory Type.

RavenSystem, 2019-11-17:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#type-3-stateless-button

vsevolod-volkov, 2019-11-17:
Solved. Thanks a lot.

closed 2019-11-17 by vsevolod-volkov

RavenSystem, 2019-11-17:
@vsevolod-volkov You are welcome. Your logs helped a lot.