RavenSystem/esp-homekit-devices #615: Wemos HAA feature request

Alexivia, 2019-11-17:
Hi,

I've experimented with this software a while back it worked very nice on my Sonos device. I also have a Wemos D1 mini device which was used at the time to experiment the software and the JSON string before flashing it to the Sonos permanently, and have since bought a couple of shields for the D1 mini.

Wemos has a couple of software written for their [RGB shield](https://github.com/wemos/D1_mini_Examples/blob/master/examples/04.Shields/WS2812B_RGB_Shield/simple/simple.ino), as well as for their [Temperature and Humidity shield](https://github.com/wemos/WEMOS_SHT3x_Arduino_Library). Is there any change these can be somehow integrated in the HAA software? They would be a very cool addition as the Wemos boards are very small and can be easily integrated in a lot of custom projects!

Regards, and keep up the good work!

mafyata, 2019-11-17:
I have wemos d1 mini and dht22 sensor. I use haa to add temp and humidity to apple home app.

mafyata, 2019-11-17:
It works on nodemcu too.

Alexivia, 2019-11-17:
> I have wemos d1 mini and dht22 sensor. I use haa to add temp and humidity to apple home app.

Yes, that sensor type is already supported according to this project's wiki, however the SHT30 (mentioned in the link I provided) is not... But yes, some of their shields might already be supported.

RavenSystem, 2019-11-18:
Sorry @Alexivia . Only PWM RGB drivers are supported and I am not interested in adding more type because a PWM driver is very cheap and small, and I prefer to focus in adding other features, like IR commands. Same thing with SHT30. You can buy a DHT22 for less than $3, and it is small, but if you want, you can buy a DHT11 that is even smaller and cheaper.

Alexivia, 2019-11-27:
While I can definitely understand you point on the atmospheric sensor, I will strongly disagree on the LED driver... A PWM driver is only suitable if either the LED string is small, or the driver itself has a high power switching circuitry that can drive a large string of LEDs... otherwise, addressable LEDs such as the ones found on the mentioned RGB shield are the way to go for large LED strings, and even if you desire multi-color strings with patterns being supported as well...

RavenSystem, 2019-11-27:
Shelly RGBW2 is a PWM driver that supports up to 288W, and only costs 18€.

closed 2019-11-29 by RavenSystem