RavenSystem/esp-homekit-devices #618: Feature request: again regarding bistable pulse relays and HAA

vsevolod-volkov, 2019-11-18:
Hi Jose,
In continuation of #550 and #577.

I have an idea how to adopt bistable pulse relay to HomeKit and not to break HomeKit paradigm.
It is possible to implement condition for checking input GPIO before applying change to output GPIO like in example shown below:
```
            "0": {
                "r": [
                    {
                        "c": {
                          "g": 5,
                          "p": 0
                        },
                        "g": 4,
                        "v": 1,
                        "i": 0.1
                    }
                ]
            },
```
This example assumes the "c" (condition) element has the same subelement meaning as described for button.

This approach has possible additional benefit: consumer gets the ability to program some kind of "smart" actions like "set output GPIOn if input GPIOx is on and set output GPIOm, when input GPIOx is off".

_Originally posted by @vsevolod-volkov in https://github.com/RavenSystem/esp-homekit-devices/issues/577#issuecomment-554775157_

RavenSystem, 2019-11-18:
No, sorry. Actually this is very hard to code because the base code of HAA must be changed. And you can do the same using additional HomeKit switches and automations.

vsevolod-volkov, 2019-11-24:
Hi Jose,

I just finished developing and debugging that feature. It works well.

Please refer to just added commit at: [https://github.com/vsevolod-volkov/esp-homekit-devices/commit/e0d3e16dde85ba4dc867c6bdb4ccd2b236580ce6](url)

I tried to completely follow your coding style as I can.
The current functionality intervention is minimal, the only executable statement I added to your code is:
```
            if (!hkc_check_action_conditions(json_relay))
               continue;
```
That code is completely secure to others who do not use action conditions because new added hkc_check_action_conditions(...) function returns true by default when there are no conditions set for action.

I will appreciate you if you will merge this to your master branch.
Thank a lot.

vsevolod-volkov, 2019-11-25:
Hi Jose,

I just finished developing and debugging that feature. It works well.

Please refer to just added commit at: [https://github.com/vsevolod-volkov/esp-homekit-devices/commit/b4aabac92ead1e908f478040c026901e654c78dd](https://github.com/vsevolod-volkov/esp-homekit-devices/commit/b4aabac92ead1e908f478040c026901e654c78dd).

I tried to completely follow your coding style as I can.
The current functionality intervention is minimal, the only executable statement I added to your code is:
```
            if (!hkc_check_action_conditions(json_relay))
               continue;
```
That code is completely secure to others who do not use action conditions because new added hkc_check_action_conditions(...) function returns true by default when there are no conditions set for action.

I will appreciate you if you will merge this to your master branch.
I hope that will enrich your project with (another))) powerful universal feature.
Thank a lot.

RavenSystem, 2019-12-06:
I have seen your code. And I can not add it because you read gpio status directly without any filter option and without using interrupts. And that is not good. For example, you can declare an array of digital inputs to use filtered GPIOs, and then use callback functions to change status of elements of array. Then, when condition is checked, you must read array, and not gpio. And other issue is that when condition is not true, state accessory changes, showing a false state, so it is necessary to implement an additional system to control that. You can see how "kill switches" feature works. Other question is about condition over what actions, because there are 2 action types ("r" digital outputs, and "m" managed accessories), and other possible case is using condition over accessory instead actions. As I said, it is not easy to implement it in a good way... sorry.

closed 2019-12-06 by RavenSystem