RavenSystem/esp-homekit-devices #620: Please help config

skop1986, 2019-11-19:
Please help me, I can't. What config should be for sonoff 4ch pro to make 2 switches, a socket and a thermostat? I've tried everything but I lack experience. Thanks

GPL71, 2019-11-19:
The follow should be your request
4ch Pro R2:  
- setup (Hold gpio 0 for more than 10 seconds to enter setup)
-2 switches  
-1 outlet, 
-1 thermostat --->DHT22 on GPIOXX (XX check in internet but should be GPIO 2)
{"c":{"l":13,"i":1,"b":[{"g":0,"t":5}]},
"a":[
{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},
{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},
{"t":2,"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},
{"t":21,"s":5,"g":XX,"0":{"r":[{"g":4}]},"1":{"a":0},"3":{"r":[{"g":4,"v":1}]},"5":{"a":0}}

Check ... should be works

skop1986, 2019-11-19:
@GPL71 It seems that there is an error somewhere, this config does not work

GPL71, 2019-11-19:
Did you replaced XX with a valid gpio?

skop1986, 2019-11-19:
@GPL71  Sorry, I made a mistake. config is correct. Thanks A Lot! It remains to understand where are GPIO2 :)

GPL71, 2019-11-19:
Check in internet. I’ll do same

skop1986, 2019-11-19:
Thanks, everything worked out!

closed 2019-11-19 by skop1986