RavenSystem/esp-homekit-devices #621: Shelly RGBW dimming LED light

kompak23, 2019-11-19:
Hi,
is it possible to use RGBW as PWM dimmer for LED lamp? 
I'll use only one channel white, and I want to set Brightness to 50% during night. Is it possible in automation? 
Thanks for reply.
Regards,
kompak23

RavenSystem, 2019-11-20:
Yes, you can. You can setup up to 4 individual white channels.

kompak23, 2019-11-20:
Thank, but how to set it up in Homekit? Is it possible, that this accessory will be visible as 4 light points? And every of these 4 light brightness can we set from automation?

RavenSystem, 2019-11-26:
> Thank, but how to set it up in Homekit?

Like any other Homekit accessory.

> Is it possible, that this accessory will be visible as 4 light points?

Yes.

> And every of these 4 light brightness can we set from automation?

Yes.

You will see 4 individual HomeKit lightbulbs.

closed 2019-11-26 by RavenSystem