RavenSystem/esp-homekit-devices #623: T0 EU 2C Stops responding after a few moments.

batista16, 2019-11-19:
Hello, trying to setup a T0 2 gang and it stops responding after a few moments, both from homekit and to the fisical switches. 
I'm using this json: 
`{
	"c": {
		"l": 13,
		"b": [
			{
				"g": 1
			},
			{
				"g": 9
			}
		]
	},
	"a": [
		{
			"0": {
				"r": [
					{
						"g": 12
					}
				]
			},
			"1": {
				"r": [
					{
						"g": 12,
						"v": 1
					}
				]
			},
			"b": [
				{
					"g": 1
				}
			]
		},
		{
			"0": {
				"r": [
					{
						"g": 5
					}
				]
			},
			"1": {
				"r": [
					{
						"g": 5,
						"v": 1
					}
				]
			},
			"b": [
				{
					"g": 9
				}
			]
		}
	]
}`

And this is the pinout (according to the internet): 

> Button 1 - GPIO01
> Button 2 - GPIO09
> Relay 1 and LED - GPIO12
> Relay 2 and LED - GPIO05
> LED - GPIO13

Could you check my json? and it's correct maybe add it to the User Templates section. 

Also, could I suggest to add a "Logs" window to the web interface? maybe we could sew what's wrong with it.

bj-tracer, 2019-11-20:
You no need to declare buttons at init config part of json string.
It need  if you want to enter setup mode after device starts with this buttons.

So your string must be:
`{ "c": { "l": 13}, "a": [ { "0": { "r": [ { "g": 12 } ] }, "1": { "r": [ { "g": 12, "v": 1 } ] }, "b": [ { "g": 1 } ] }, { "0": { "r": [ { "g": 5 } ] }, "1": { "r": [ { "g": 5, "v": 1 } ] }, "b": [ { "g": 9 } ] } ] }`

With enter to setup mode (hold after power on):
`{ "c": { "l": 13, "b":[{"g":1,"t":5}]}, "a": [ { "0": { "r": [ { "g": 12 } ] }, "1": { "r": [ { "g": 12, "v": 1 } ] }, "b": [ { "g": 1 } ] }, { "0": { "r": [ { "g": 5 } ] }, "1": { "r": [ { "g": 5, "v": 1 } ] }, "b": [ { "g": 9 } ] } ] }`

batista16, 2019-11-20:
Thanks! that helped somewhat, but only one of the fisical buttons it's working. I'm guessing the GPIO mapping that I've found online is incorrect. Does anyone know the correct GPIO for Sonoff T0EU2C?

batista16, 2019-11-20:
Yeah so apparently the correct GPIO for the buttons are GPIO0 and GPIO9, so this is the final JSON: 
`{
	"c": {
		"l": 13,
		"b": [
			{
				"g": 0,
				"t": 5
			}
		]
	},
	"a": [
		{
			"0": {
				"r": [
					{
						"g": 12
					}
				]
			},
			"1": {
				"r": [
					{
						"g": 12,
						"v": 1
					}
				]
			},
			"b": [
				{
					"g": 0
				}
			]
		},
		{
			"0": {
				"r": [
					{
						"g": 5
					}
				]
			},
			"1": {
				"r": [
					{
						"g": 5,
						"v": 1
					}
				]
			},
			"b": [
				{
					"g": 9
				}
			]
		}
	]
}`

Feel free to add it to the user templates. :)

closed 2019-11-20 by batista16