RavenSystem/esp-homekit-devices #624: How to interlock relays

GPL71, 2019-11-21:
Hi josè ....
I have 3 relays to manage as follow:
Relay 1 ON, Relay 2, 3 OFF
Relay 2 ON, Relay 1, 3 OFF
Relay 3 ON, Relay 1, 2 OFF
- initial state must be always Relay 1ON, Relay 2, 3 OFF
I'll use a stateless to switch between conditions up

Is it possible manage it in the json config? Or an automation is necessary?

Thank you

RavenSystem, 2019-11-26:
Actually, you can interlock relays, but I think you want to interlock HomeKit switches (then each switch manages its relay), no?

GPL71, 2019-11-26:
mumble .... mumble .... that's correct josè .... relays and homekit switches too.
Is it possible?

RavenSystem, 2019-11-28:
Yes, with last version of HAA: https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#m-accessory-manager-actions-array-of-orders-attached-to-an-action-to-manage-other-accessories-from-same-device

GPL71, 2019-11-28:
Thank you very much .... sounds good ear now it’s a feature

closed 2019-11-29 by RavenSystem