RavenSystem/esp-homekit-devices #625: Shelly1 as Door opener

fIctioN600, 2019-11-21:
I am not quite sure if the json does what I want or if its even possible.

My idea is to use the shelly1 to recognize my doorbell and control the dooropener. Therefor I want that the shelly becomes a HomeKit accessory with a switch to control GPIO4 for the opener and a fake motion sensor wich is actually an external toggle connected to GPIO5. So if I press the external toggle (my doorbell) homekit says motion detected. Then I press the switch in homekit for the opener and the door opens.

JSON:
{"a":[{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":1,"s":0,"i":1.5},{"t":12,"s":0,"f1":[{"g":5,"t":1}],"f0":[{"g":5,"t":0}],"i":1}]}

RavenSystem, 2019-11-26:
Try this JSON. It is more adequate to your purpose (lock mechanism + occupancy sensor):

`{"a":[{"t":4,"i":4,"0":{"r":[{"g":4,"v":1,"i":1.5}]}},{"t":6,"i":10,"f1":[{"g":5,"t":1}]}]}`

closed 2019-11-29 by RavenSystem

Kienz, 2019-12-09:
@fIctioN600 Can you post a picture of installation? I don't know how to connect it.