RavenSystem/esp-homekit-devices #626: Sonoff TH16 stopped showing humidity after 0.7.2

than-sad, 2019-11-22:
Hi, I just updated my Sonoff TH16 to firmware 0.7.2 and now the humidity indicator under setting disappeared.  Is it a normal behavior?

Im using the JSON from templates: {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":21,"s":5,"g":14,"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0}}]}

Also the physical button doesn't on or off the thermostat, probably adding the button in the JSON will solve it but im not really know where should I type it.

Best regards,
Than.

RavenSystem, 2019-11-23:
Apple removed Humidity characteristic from new HomeKit Thermostat version. You can add it with an additional accessory in your JSON.

You must add your physical button in your JSON.

than-sad, 2019-11-23:
Great! Thank's for the info. I will try to learn about JSON correct sintaxis.

Regards.
Than.

RavenSystem, 2019-11-26:
I personally use that method with my thermostats.

closed 2019-11-26 by RavenSystem

than-sad, 2019-11-27:
Hi, I was trying to add spaces to the TH16 template, just to help me understand and see where I should add code but also using Sublime text editor (in Javascript - JSON file) I found it a little hard:
{
 "c": {
   "l": 13,
	"b": [
      {
      	"g": 0,
      	"t": 5
	  }
		 ]
	  },
		"a": [
		  {
			"t": 21,
			"s": 5,
			"g": 14,
			"0": {
				"r": [
					{
					"g": 12
					}
					]
				},
				"1": {
				"a": 0
				},
				"3": {
					"r": [
						{
						"g": 12,
						"v": 1
					}
				]
				},
				"5": {
				"a": 0
			}
		}
	]
}

Can you share the code you are using please?

Best regards.

RavenSystem, 2019-11-27:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":21,"s":5,"g":14,"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0}},{"t":23,"g":14}]}`

than-sad, 2019-11-28:
Thanks José