RavenSystem/esp-homekit-devices #628: Disable Emergency Setup Mode

pikrato, 2019-11-22:
Is there a way to disable the "emergency setup mode" to avoid having to reconfigure all my devices everytime someone pulls the plug? This would be very helpful since every time I cut the main switch to install a new relay, afterwards I have to reconfigure all my switches when I turn the electicity back on.

mafyata, 2019-11-22:
It enter in setup mode if you disconnect power , connect and within 3 or 5 seconds disconnect and connect again.

pikrato, 2019-11-23:
As far as I know that‘s how you go into “emergency setup mode”. I want to avoid that. 
Let’s say I’m planning on changing the switch in a floor lamp in my living room to a shelly switch. I don’t want to have to setup the switch every time someone pulls the lamp from the plug. 
I noticed that there is a variable in the HAA main.c file called “EXIT_EMERGENCY_SETUP_MODE_TIME” and it’s set to 3000. Would it be disabled if I changed it to 0 or at least reduced the number to like 60 seconds?

RavenSystem, 2019-11-26:
When device enters in setup mode, none setting is changed or removed. You must not reconfigure anything.

I am not going to disable emergency setup. It is very difficult to activate it accidentally. 3000 are milliseconds, not seconds. To enter setup mode using emergency method, you must power off your device, power on and immediately power off again (you have less that 3 seconds to do this), and power on it again. In that moment, device will enter in setup mode.

closed 2019-11-26 by RavenSystem