RavenSystem/esp-homekit-devices #629: json configuration for Sonoff "pulse switch"

Powerbook001, 2019-11-25:
Hallo,
Is there a chance to configure switch to behave as a pulse button (1 second ON an then OFF?)
I tried using "inching" and works but not correctly,  once I use it on HomeKit, the relay will work correctly and closes the contact for 1 second and then rest, but I will have to press the button twice in order to turn to the OFF status of the button. 

This is the Json I use with my Sonoff 4ch pro:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1,"i":1}]},"b":[{"g":14}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1,"i":1}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1,"i":1}]},"b":[{"g":10}]}]}

Thank you

bj-tracer, 2019-11-26:
Hi!
You no need to use "off" action if you using "inching", I think.

Powerbook001, 2019-11-26:
The Sonoff 4 ch is working correctly, only that Homekit does not realize that she switch is off untill I press for a second time the button.

RavenSystem, 2019-11-26:
@Powerbook001 You must set inching at the root of the accessory, not into relay action.

Powerbook001, 2019-11-26:
Thank you !! that worked greatly.
Bye the way, great jib with RavenSystem!!

closed 2019-11-26 by RavenSystem