RavenSystem/esp-homekit-devices #631: Magic Home RGB+CCT White Control

brymonster, 2019-11-27:
[This](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#magichome-rgbw-led-strip-controller-r-pin-15-g-pin-13-b-pin-12-w-pin-14) template gives the impression you can only have RGBW. Not WW or CW. 
Curious if HAA has the ability to 'tune' white with a controller and LEDs that have Warm and Cool White. Like [this magic home controller.](https://www.aliexpress.com/item/32792842998.html?spm=2114.search0104.3.1.cbd5238bNyBLJr&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10068_10130_10547_319_317_10548_10696_453_10084_454_10083_10618_431_10139_10307_537_536_10059_10884_5736015_10887_100031_321_322_10103_5736115,searchweb201603_51,ppcSwitch_0&algo_expid=864d5028-ab30-4ffd-9f46-0a3fca0cd85e-0&algo_pvid=864d5028-ab30-4ffd-9f46-0a3fca0cd85e&transAbTest=ae803_4) Has anyone had success with this?

Bene2103, 2019-11-27:
If you take a look at the Wiki you see that you can control a RGBW led strip, no WW and CW at the moment.

But you can tune your color temp in the Home App if you select temperature not a color.

brymonster, 2019-11-27:
Thanks @Bene2103! 
I did take a look at the wiki, as linked in the op. 
So ‘no WW and CW at the moment’, but it DOES work in the Home app (the only place that matters)?.. that’s working to me. 
What device are u using?

Bene2103, 2019-11-27:
Yep it works. Maybe not as good as dedicated WW and CW but it’s enough for me. If you want to I can try taking some pics of WW CW and in between. 

Im using a „Magic Home RGB WiFi“ From Amazon (don’t wanted to wait 10 weeks for them to arrive ;-) ) And also a ws2812 RGB Matrix in a 3D printed lamp.

brymonster, 2019-11-27:
Pics would be awesome! 
Plus I wanna see this 3D lamp. 🧐

closed 2019-11-29 by RavenSystem

Bene2103, 2019-11-29:
Sorry for the late reply. On the photos the color is a little bit off. It looks definitely not that blueish and the warm color is a bit more warm.

Cold White
[image]

Warm White
[image]

The Lamp itself
[image]
[image]

kevinjohncutler, 2020-09-06:
@brymonster were you able to get this working with that specific controller? If so, did you have to flash it wired or is there a way to do it OTA?

kevinjohncutler, 2020-09-09:
@ Bene2103 what json did you use for your LED controller? I tried the Magic Home RGBW json from the database, and as I would expect it doesn't utilize the cold white LED at all, but worse it seems not to be able to do pure colors (specifically red). I'm using the same controller that @brymonster linked at the top post.

Bene2103, 2020-10-01:
The pictures i showed in this thread was only to show my 3d printed lamp as requested by brymonster. This lamp uses a different software to work with ws2812b adressable leds.

So no RGBWW or RGBCW at all. This was just to show that you can also get CW an WW in an RGB only device.