RavenSystem/esp-homekit-devices #632: Proximity sensor

Powerbook001, 2019-11-30:
I'm having difficulty on trying to make the proximity sensor connected to GPIO 14 work in connection with the relay. 
Can anybody tell me where is my mistake?

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14}],"f1":[{"g":14}]}]}

Thank you.

RavenSystem, 2019-11-30:
You have a switch to manage GPIO 12, and contact sensor. Now, you have 2 possibles scenarios:

a) Manage gpio12 directly with the contact sensor, or
b) Manage Homekit switch with the contact sensor.

What do you want?

Powerbook001, 2019-11-30:
Could you please be so kind to have both examples?
Thank you.

RavenSystem, 2019-11-30:
a) `{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14}],"f1":[{"g":14}],"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]}}]}`

b) `{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14}],"f1":[{"g":14}],"0":{"m":[{"g":1}]},"1":{"m":[{"g":1,"v":1}]}}]}`

Powerbook001, 2019-11-30:
Thank you José,
I'm understanding now how the JSON configuration works, little by little.  ;)

Powerbook001, 2019-11-30:
Sorry Josè, I tried the a) and works, but once the proximity contact on GPIO 14 is closed, it does not open anymore also if there is no proximity. The relay works fine.
Should I add some parameters?
Regards

Powerbook001, 2019-12-03:
Can you still give me a hint?
Thank you

Nico

RavenSystem, 2020-01-10:
There was an error in my JSON: I missed "t":0

a) `{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14,"t":0}],"f1":[{"g":14}],"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]}}]}`

b) `{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14,"t":0}],"f1":[{"g":14}],"0":{"m":[{"g":1}]},"1":{"m":[{"g":1,"v":1}]}}]}`

closed 2020-01-10 by RavenSystem