RavenSystem/esp-homekit-devices #636: Ota Updates with Shelly

sancheztineo, 2019-12-04:
I have two types of flashed devices, Sonoff and Shelly. With the Sonoffs I can update OTA update . With the Shelly doing the same, the web view option does not appear (Only appear reset homekit ID) ,  I can't OTA update.

What can be the motive?

RavenSystem, 2019-12-06:
How did your flash your devices?

sancheztineo, 2019-12-08:
I flash my devices with Python with cmd in windows 10.
With this commands:
"esptool.py --port COM3 --baud 115200 erase_flash"
"esptool.py --port COM3 --baud 115200 write_flash -fs detect -fm dout 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin"

When I flash Sonoff devices all is OK (Sonoff mini Sononff Basic R1 and R3, and Sonoff TH).
When I flash Shelly 2.5 and Shelly 1, devices install te last version of firmware, but only the first time. In the web Wiew don`t show the options OTA Updates.

RavenSystem, 2019-12-08:
This is the problem `-fs detect`. You changed value. See right command from installation instructions in the Wiki.

closed 2019-12-08 by RavenSystem

sancheztineo, 2019-12-09:
Thank you very much, José Antonio.