RavenSystem/esp-homekit-devices #637: Shelly RGBW2 with RGB LED

macjl, 2019-12-04:
Hello,

I'm using HAA on a Shelly RGBW2 with a RGB led strip.
Here is my JSON :
{"a":[{"t":30,"d":0.5,"r":12,"g":15,"v":14,"b":[{"g":5,"p":0}]},{"t":3,"f0":[{"g":3,"t":1}],"f1":[{"g":3,"t":2}],"f2":[{"g":3,"t":3}]}]}

When setting white colour and brightness to 100% in Home App, the LED are not at 100% of their power.

I tried to add a 'factor' option, but when doing that, the LED are now at 100% of their power when setting to 30% in Home App, whatever the factor value.

I also have a color issue. The color are whiter than what is set in Home App.

macjl, 2019-12-04:
While watching the code, I found this : 

`if (cJSON_GetObjectItem(json_context, LIGHTBULB_FACTOR_R) != NULL) {
            lightbulb_group->factor_r = (float) cJSON_GetObjectItem(json_context, LIGHTBULB_PWM_GPIO_R)->valuedouble;
        }`

Shouldn't it be this instead?
`if (cJSON_GetObjectItem(json_context, LIGHTBULB_FACTOR_R) != NULL) {
            lightbulb_group->factor_r = (float) cJSON_GetObjectItem(json_context, LIGHTBULB_FACTOR_R)->valuedouble;
        }`

Same thing for Blue and Green.

RavenSystem, 2019-12-04:
Ops!

RavenSystem, 2019-12-04:
Update to v0.7.9, please.

macjl, 2019-12-04:
Version updated. It works great for factor values.

But the colour are still not accurate between what is set on Home App, and what LEDs are showing.

mafyata, 2019-12-04:
I think that this is from led controllers. I have 2 magic home led controllers, exactly the same , but one is not showing accurate colors.

macjl, 2019-12-05:
I have remove and reset, and re-add the device to homekit. The colours are better now.

The first time I added it, the JSON string was set up as RGBW. I changed it after to RGB, but maybe Homekit keeped some parameters in cache?

RavenSystem, 2019-12-05:
It is strange, because HomeKit does not know how your device works. HomeKit only sends color, and then device must show color using its channels.

macjl, 2019-12-05:
Ok. I will make more test when back home.
The tests I made yesterday seems better, but this morning, I still obtain random colours that not correspond to what I asked.

Could the "Custom PWM Frequency" have an impact? I've set it to 65535 to reduce noises, and don't really know what's the impact on the device and performances.

macjl, 2019-12-05:
Here are my tests : 

Asking red, get pink : 
[image]

Asking blue, get blue : 
[image]

Asking green, get blue : 
[image]

Asking bondi blue, get bondi blue : 
[image]

I can obtain real green, red and blue, but I have to be exactly on the good colour in Home App. If I move 1px left or right, the colour change dramatically.

RavenSystem, 2019-12-06:
Homekit uses brightness, hue and saturation to set color, but I need RGBW values. For RGB and RGW I use this algorithm (url source is in the code): https://github.com/RavenSystem/esp-homekit-devices/blob/master/devices/HAA/main.c#L995

If you know a better algorithm to convert HSI to RGBW, please, tell me it, because I have a color lightbulb and it is crazy setting red or green colors.

RAZRu, 2019-12-09:
Something wrong with colors with variable "v" in code
Magichome RGB controller

1. When I use This JSON:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":5,"g":12,"v":13,"b":[{"g":0}]}]}:
Red is Red (r)
Green is Blue (g) 
Blue is Pink (v)

When I use This JSON: {"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":5,"g":13,"v":12,"b":[{"g":0}]}]}
Red is Red (r)
Green is Green (g)
Blue is Yellow (v)

In RavenSystem firmware everything worked well

RavenSystem, 2019-12-09:
Try v0.8.2.

macjl, 2019-12-09:
Much better! Thank you for finding it!

closed 2019-12-09 by macjl

RAZRu, 2019-12-09:
> Try v0.8.2.

Now Works perfect, thank you!