RavenSystem/esp-homekit-devices #638: Support for R,G,B,CW,WW bulbs

danielhelmstedt, 2019-12-05:
Hi, is it possible to set up a bulb that features both RGB and customisable colour temp?

This is the one I'm trying to set up.
https://blakadder.github.io/templates/arlec_GLD122HA.html

RavenSystem, 2019-12-05:
RGB-CW-WW is not supported yet, sorry.

danielhelmstedt, 2019-12-07:
Okay no worries, I like the sound of that 'yet' :P

RavenSystem, 2019-12-09:
I only need an algorithm that converts HSI values to R G B CW WW.

magichew, 2019-12-10:
Hi all

I asked for similar on the for a lightstrip controller here. Addressing 5 colours would be really useful. 

https://github.com/RavenSystem/esp-homekit-devices/issues/504

peros550, 2020-01-31:
Any updates here?

magichew, 2020-02-25:
Just jumping on this to keep it alive. I haven't been able to find an algorithm to facilitate this but the bulbs and the RGB strip controller that I have had full RGBCWWW support via Tasmota so I assume it will be possible.

rasod, 2020-02-25:
I cobbled some code together which seems to work at taking RGB in and outputting 5 values. The code is complete garbage (and in JS) but is a starting point on what needs to be done. Will post soon.

rasod, 2020-02-25:
Here is some very rough and verbose code, takes a RGB value and converts to RGB CW WW. I’m not going to pretend I know exactly what is going on since I couldn’t find a well documented implementation. This was mostly from homebridge-mqttthing. ESPHome has an implementation but it was hard to exact just the relative code. 

https://gist.github.com/rasod/42eab9206e28ca91c8d9f926fa71a938

It’s JavaScript, since I don’t know C++ (some might argue I don’t know JS after seeing this code) but it is all pretty simple math. Just trying to help.

brymonster, 2020-03-08:
@rasod tha me for your research.  We will have to wait and see if it bears fruit. 
I’m also looking forward to RGB-CCT support.

rasod, 2020-04-13:
Just tried out the new firmware - thanks. It works better you can turn the bulb on/off now and get “most” of the colours.

Some colours don’t appear correctly, not sure if it is my configuration or the algorithm I shared.

Is there a way to view a log or console, to see the input/output values over the network? I can’t get to any of the programming pins on the bulbs I have.

Thanks Again

RavenSystem, 2020-04-14:
> Is there a way to view a log or console, to see the input/output values over the network? I can’t get to any of the programming pins on the bulbs I have.

No, only UART.

I understand that you have a 5 channel RGBWWCT device. Can you test colors with only RGB channels (no whites)?

rasod, 2020-04-14:
The colours seem to work fine.

However once you start to mix in the whites the colours get lost and the white overpowers the colour.

RavenSystem, 2020-04-14:
That is normal, because white channels are more powered than colours in many devices. You can use factors to increase colors, or decrease whites.

rasod, 2020-04-14:
That is what I expected, been trying to adjust the factors but can’t seem to get the bulb back into setup mode. AP will show up then disappear before time to connect or get error connecting. Without a physical button or access to the pins its a real PITA.

For the colour temperature it currently works that when you pick warm white (12 o’clock on the Home slider) the warm white led is on and the RGB is mixing trying to match. When you pick cold white (6 o’clock on the slider) the cold white is on and RGB is mixing trying to match. I don’t have any “official” RBGCCT bulbs so I don’t know what should be expected.

RavenSystem, 2020-04-14:
@rasod You can get setup mode turning on and off quickly your bulb8 times from HomeKit, or using emergency setup mode:
- Power cut.
- Power on.
- Wait 1.5-2 seconds.
- Power cut.
- Power on and device will enter in setup mode

Main problem with algorithm is that does not mix CW and WW for middle values. Color accuracy is other matter.

rasod, 2020-04-14:
I have tried both methods many times. The Access Point will appear and when I try to connect I don’t get the captive screen, the AP disappears and the bulb restarts. Not sure if it times out or crashed.

Sorry about the algorithm I couldn’t find well documented one.

RavenSystem, 2020-04-14:
Your device is in setup mode, but when it connects to your wifi, AP disappears. You must use device hostname or IP address:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

Don’t worry about algorithm, it is a good stater point ;-)

rasod, 2020-04-14:
doh!🤦‍♂️ I knew that, not sure why I kept trying the AP. thanks

closed 2020-05-15 by stale[bot]

kevinjohncutler, 2020-09-06:
So, what is the status on RGB-CW-WW support?

toshibochan, 2020-09-06:
Here is good example for the code.
https://github.com/maccoylton/esp-homekit-rgbw-strip