RavenSystem/esp-homekit-devices #639: Proximity sensor does not change state

Powerbook001, 2019-12-05:
Hallo Josè,

IN The following examples that you kindly posted, I cannot make work the proximity sensor.
Once it has the first contact, for this works, it does not change state anymore. The sonoff unit flashes every time the proximity is engaged and disengaged, but HomeKit does not change the state in the App.
Is there still something I must do?
Thank you again.

Nico

a) {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14}],"f1":[{"g":14}],"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]}}]}

b) {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"t":5,"f0":[{"g":14}],"f1":[{"g":14}],"0":{"m":[{"g":1}]},"1":{"m":[{"g":1,"v":1}]}}]}

RavenSystem, 2019-12-06:
It is normal, because you are using the same input for both sensor states: "f0":[{"g":14}],"f1":[{"g":14}]

Maybe you want: "f0":[{"g":14,"t":0}],"f1":[{"g":14}]

Powerbook001, 2019-12-06:
Hallo,
I Tried you hint, but now in both configuration the Relay is driven by the contact sensor, and is not indipendent as a simple switch driven from Homekit.
Let's say that option a) is ok, and that option b) is now like  a)
What ma I missing? 

Thanks for your patient.

Nico

a)
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"m":[{"g":1}]},"1":{"m":[{"g":1,"v":1}]},"t":5,"f0":[{"g":14,"t":0}],"f1":[{"g":14}]}]}

b)
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0},{"0":{"m":[{"g":1}]},"1":{"m":[{"g":1,"v":1}]},"t":5,"f0":[{"g":14,"t":0}],"f1":[{"g":14}],"s":0}]}

closed 2020-01-10 by RavenSystem