RavenSystem/esp-homekit-devices #640: Feature suggest for LightBulb

macjl, 2019-12-06:
Hello José,

I'm playing with HAA on a lightbulb for some days. It's very fun, and I wanted to suggest you some feature I think that could be nice to implement.

- Option to enable/disable Autodimmer on default GPIO, and add actions to start and stop it. It will allow user to choose which GPIO/press type will trigger Autodimmer feature.
Example of what could be done whith JSON : 
```
"b":[{"g":5,"t":1}],   => Nothing changed. Lightbulb is activated/deactivated with GPIO 5.
"a":0,                => Disable Autodimmer on GPIO 5.
"f4":[{"g":3,"t":2}], => Double press on GPIO 3 will launch AutoDimmer
"f5":[{"g":1,"t":1}]  => Single press on GPIO 1 will stop it.
```
- Pre-program specific colour/luminosity that can be activated with a GPIO input.
Example of what could be done whith JSON : 
```
  "f6.100.000.000":[{"g":0,"t":2}],  => For RGB, Double press on GPIO 0 will ativate RED at 100%, GREEN at 0% and BLUE at 0%
  "f6.010.010.010":[{"g":0,"t":3}]   => For RGB, Long Press on GPIO 0 will ativate RED at 10%, GREEN at 10% and BLUE at 10%
  "f7.030.040.010.100":[{"g":0,"t":4}]   => For RGBW, Very Long Press on GPIO 0 will ativate RED at 30%, GREEN at 40%, BLUE at 10% and WHITE at 100%
```

Instead of fx.xxx.xxx.xxx, It could also be Hexadecimal value of colour, like #ff0000 for 100% red or #808080 for 50% white. 

What do you think about these suggests?

RavenSystem, 2019-12-06:
> Option to enable/disable Autodimmer...

Actually, you can manage autodimmer using a button and a switch. With your concept, autodimmer could not be managed with the same mount-wall switch used to turn on/off lightbulb.

> Pre-program specific color/luminosity...

It is possible, but not easy. You can not used "f6.100.000.000" as a label, because label must be a known string (I search the label and I read its value). It is necessary to develop an additional protocol like: "6":"ff00000000" (R G B CW WW), and then "f6":[{"g":0,"t":2}]. I had some ideas to implement it (I call it "custom scenes"), but actually I want to focus on garage door and window covers before adding new features like this. But I will add it to TODO list.

macjl, 2019-12-06:
> Actually, you can manage autodimmer using a button and a switch. With your concept, autodimmer could not be managed with the same mount-wall switch used to turn on/off lightbulb.

I understand what you mean.
For my use, I don’t need the AutoDimmer function. So an option to activate/deactivate it could be great.  

> It is possible, but not easy. You can not used "f6.100.000.000" as a label, because label must be a known string (I search the label and I read its value). It is necessary to develop an additional [...]

Thank you. I’m also interested in garage door function!
For the moment, I use a push button to have Custom Scene. The target colour/luminosity is set in HomeKit. The only limitation is that it doesn’t work in case of WiFi or Hub problem.

RavenSystem, 2019-12-06:
With last version, you can disable autodimmer setting "autodimmer step percentage" to 0.

macjl, 2019-12-06:
Thank you! I’ll try it tomorrow.

macjl, 2019-12-08:
I've disable auto dimmer, and was able to add a stateless button on same GPIO to have Custom Scene. Here is my Json for example with RGBW2 and two push button on GPIO 1 (TX) and 3 (RX) : 
`{"c":{"n":"srgbw2-haa-chp","q":16383},"a":[{"t":30,"e":0,"r":12,"g":15,"v":14,"fr":3,"fg":3,"fv":3,"b":[{"g":1,"t":1}],"f2":[{"g":1,"t":2}],"f3":[{"g":3,"t":2}]},{"t":3,"f0":[{"g":3,"t":1}],"f1":[{"g":1,"t":3}],"f2":[{"g":3,"t":3}]}]}`

Thank you! All working as expected!

closed 2019-12-09 by RavenSystem

p0we7, 2021-03-15:
> disable autodimmer setting "autodimmer step percentage" to 0.

This feature doesn't seem to work anymore, if I set `"e": 0` then the bulb will not turn on and will always be at 0% even if I click on it.

@RavenSystem