RavenSystem/esp-homekit-devices #641: Better support for CCT Bulbs

danielhelmstedt, 2019-12-07:
Hi! is it possible to get control for 2-channel CCT bulbs where one channel is a dimmer and the other controls temperature via PWM between WW and CW?

See this Tasmota template - https://blakadder.github.io/templates/arlec_GLD112HA.html

Thank you!

danielhelmstedt, 2020-01-08:
Just bumping this, any chance it’s possible?

RavenSystem, 2020-01-10:
Not possible. Only normal PWM devices are supported. That device uses PWM in a custom way.

closed 2020-01-10 by RavenSystem

danielhelmstedt, 2020-01-13:
Is this something that should be taken up with esp-homekit then? Are you getting the temperature value from the HSI or from the Mired characteristic?If you are using mired, it seems like it should be easy to implement a function that converts mired to pwm value -

assuming say 3000K/~333mired and 5700K/~175 (my bulbs in question). That’s a difference of 158, so all you need is to divide 255/158 and apply that value as a Pwm output, no?

danielhelmstedt, 2020-01-13:
I know at least some Trådfri bulbs work this way as well, so it has at least been done before