RavenSystem/esp-homekit-devices #642: Can’t enter setup mode

mafyata, 2019-12-07:
Hi , I updated to 7.10 but can’t enter in setup mode. Tried http://hol-led and http://haa-f8f4fc
I entered hol-led in hostname in json

RavenSystem, 2019-12-07:
Custom hostname doesn't work with setup mode, but haa-f8f4fc must work. Can you try `ping haa-f8f4fc` ?

mafyata, 2019-12-07:
I entered setup mode by ip : http://10.0.1.104/
ping HAA-F8F4FC
Ping request could not find host HAA-F8F4FC. Please check the name and try again.

Pinging 10.0.1.104 with 32 bytes of data:
Reply from 10.0.1.104: bytes=32 time=29ms TTL=255
Reply from 10.0.1.104: bytes=32 time=15ms TTL=255
Reply from 10.0.1.104: bytes=32 time=29ms TTL=255
Reply from 10.0.1.104: bytes=32 time=16ms TTL=255

mafyata, 2019-12-07:
I use latest apple airport extreme router. May be the problem is with the router.

RavenSystem, 2019-12-08:
What DNS server are your using?

Can you try nslookup?
`nslookup 10.0.1.104`

mafyata, 2019-12-08:
Server:  UnKnown
Address:  10.0.1.1

*** UnKnown can't find 10.0.1.104: Non-existent domain

mafyata, 2019-12-08:
Now I tried Bonjour Browser. It is not showing anything. The problem is with apple airport routers.

RavenSystem, 2019-12-08:
It is strange that an Apple Airport doesn't allow DHCP hostname option. I have an Asus router and I have not any problem.

mafyata, 2019-12-08:
I can give you access to my computer for some tests.

macjl, 2019-12-08:
Have you tried to use : 
ping HAA-F8F4FC.local

mafyata, 2019-12-08:
>ping haa-f8f4fc.local
Ping request could not find host haa-f8f4fc.local. Please check the name and try again.

N3mo72, 2019-12-17:
Sorry for off-topic but : what is the correct JSON syntax to insert the custom hostname and its position in the string?
Thank you.

mafyata, 2019-12-17:
Here is one example of my device:
{"c":{"n":"Kuhnq-Led","b":[{"g":0,"t":0}]},"a":[{"t":30,"s":5,"r":5,"g":12,"v":13}]}
Hostname is: Kuhnq-Led

N3mo72, 2019-12-17:
This is mine code but not work :
{"c":{"n":"my-switch","l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":5}]}

What's wrong ?

mafyata, 2019-12-17:
It looks ok. What is the problem?

N3mo72, 2019-12-17:
This is the Json config that use in first configuration. After reboot, i have always HAA-XXXXXX. I tried to select the option “Reset ID” but not work

closed 2020-01-10 by RavenSystem