RavenSystem/esp-homekit-devices #644: Possible heap memory leakage

vsevolod-volkov, 2019-12-08:
Hi, Jose!

I have noticed possible error in accessory initialisation code. Pleas refer to code fragment below:
```
        // Digital outputs GPIO Setup
        for (uint8_t int_action=0; int_action<MAX_ACTIONS; int_action++) {
            char *action = malloc(2);
            itoa(int_action, action, 10);

            if (cJSON_GetObjectItemCaseSensitive(json_accessory, action) != NULL) {
                if (cJSON_GetObjectItemCaseSensitive(cJSON_GetObjectItemCaseSensitive(json_accessory, action), DIGITAL_OUTPUTS_ARRAY) != NULL) {
                    cJSON *json_relays = cJSON_GetObjectItemCaseSensitive(cJSON_GetObjectItemCaseSensitive(json_accessory, action), DIGITAL_OUTPUTS_ARRAY);
                    free(action);

                    ...
                }
            }
        }
```
Here you allocate char *action on every iteration over accessory action but free it only if "r" action element exists.
I do not understand why are you using heap allocation here. Why not to use automatic variable (char action[2];)?

vsevolod-volkov, 2019-12-08:
Why not to use automatic variable as char action[2]?

RavenSystem, 2019-12-08:
I did it. It is fixed. Thanks!

closed 2019-12-08 by RavenSystem