RavenSystem/esp-homekit-devices #645: Impossible to enter in setup mode

PNM96, 2019-12-08:
Hello! 
After updating to HAA v0.7.10, I try to enter in setup mode (I toggle quickly the switch 8 times) but without result... 
the HAA network appears for a few seconds .... but it is impossible to connect to it.  Thean shortly after, the accessory “no responds”. 

What could be the problem?

RavenSystem, 2019-12-08:
From the Wiki:

PNM96, 2019-12-08:
thanks! I tried to connect to the hostname, but it doesn't work.

mafyata, 2019-12-08:
Try with ip address.

brymonster, 2019-12-11:
Same issue here. No HAA-XXX network ever showing up. 
Tried toggling more than 8 times fast & slow. 
Its on the [network!](https://user-images.githubusercontent.com/12586478/70642704-b086d380-1c0d-11ea-8fc0-c05b8a8deff0.jpeg)

```
iMac:~ brymonster$ ping haa-b7c3ed.local
PING haa-b7c3ed.local (192.168.1.15): 56 data bytes
64 bytes from 192.168.1.15: icmp_seq=0 ttl=255 time=28.100 ms
64 bytes from 192.168.1.15: icmp_seq=1 ttl=255 time=21.938 ms
64 bytes from 192.168.1.15: icmp_seq=2 ttl=255 time=9.291 ms
64 bytes from 192.168.1.15: icmp_seq=3 ttl=255 time=17.761 ms
64 bytes from 192.168.1.15: icmp_seq=4 ttl=255 time=29.155 ms
64 bytes from 192.168.1.15: icmp_seq=5 ttl=255 time=40.939 ms
64 bytes from 192.168.1.15: icmp_seq=6 ttl=255 time=35.474 ms
64 bytes from 192.168.1.15: icmp_seq=7 ttl=255 time=10.757 ms
^C
--- haa-b7c3ed.local ping statistics ---
8 packets transmitted, 8 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 9.291/24.177/40.939/10.587 ms
iMac:~ brymonster$ ping 192.168.1.15
PING 192.168.1.15 (192.168.1.15): 56 data bytes
64 bytes from 192.168.1.15: icmp_seq=0 ttl=255 time=13.908 ms
64 bytes from 192.168.1.15: icmp_seq=1 ttl=255 time=12.256 ms
64 bytes from 192.168.1.15: icmp_seq=2 ttl=255 time=164.790 ms
64 bytes from 192.168.1.15: icmp_seq=3 ttl=255 time=117.161 ms
64 bytes from 192.168.1.15: icmp_seq=4 ttl=255 time=114.602 ms
64 bytes from 192.168.1.15: icmp_seq=5 ttl=255 time=103.868 ms
64 bytes from 192.168.1.15: icmp_seq=6 ttl=255 time=11.409 ms
^C
--- 192.168.1.15 ping statistics ---
7 packets transmitted, 7 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 11.409/76.856/164.790/58.469 ms
```

mafyata, 2019-12-11:
What version of haa are you using? With new version is changed the way to enter settings in setup mode. You must enter ip address/hostname of your haa device.

PNM96, 2019-12-11:
I solved by connecting to the ip address.  with the hostname it didn't work

brymonster, 2019-12-11:
HAA .8.2
Yes. I have tried using multiple browsers. 
Both http://www.haa-b7c3ed and http://192.168.1.15

How long do you have to wait after the 8 toggle dance for setup mode to start?

brymonster, 2019-12-11:
I just keep getting sent:
http://www.dnsrsearch.com/search/?q=http://www.haa-b7c3ed/

mafyata, 2019-12-11:
When you see in home app that you can’t controll the device , wait 2-3 secs and open browser.

brymonster, 2019-12-11:
ohh. I didnt realize that was an indication of Setup Mode! I will keep trying!

brymonster, 2019-12-11:
I just toggled it prob 100 times. Home app never lost control.

brymonster, 2019-12-11:
Success. holy moly. thanks guys. 
Wow, that Setup Mode. Achievement unlocked.

closed 2019-12-13 by RavenSystem