RavenSystem/esp-homekit-devices #647: Can't complete the Setup. No web appears

Shimakuchi, 2019-12-08:
Hi, I followed the instructions from [here](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation). I am struggling with point 4. I can connect to the **LCM-xxxxxx** network, but there appears NOTHING. Not on my iPhone, iPad nor on any MacBook or even Windows Laptops I have tried. I get a weird IP-Address like 169.254.210.29/255.255.0.0

Is there any other way to get to the WebUI to complete the setup?

The device I use is a Shelly 2.5. What I did so far:

`esptool.py -p /dev/cu.usbserial-00000000 erase_flash`

> esptool.py v2.8
> Serial port /dev/cu.usbserial-00000000
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: ec:fa:bc:c4:7a:c8
> Uploading stub...
> Running stub...
> Stub running...
> Erasing flash (this may take a while)...
> Chip erase completed successfully in 0.0s
> Hard resetting via RTS pin...

`esptool.py -p /dev/cu.usbserial-00000000 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \`

> 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
> esptool.py v2.8
> Serial port /dev/cu.usbserial-00000000
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> Crystal is 26MHz
> MAC: ec:fa:bc:c4:7a:c8
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Flash params set to 0x0320
> Compressed 3104 bytes to 2169...
> Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 119.5 kbit/s)...
> Hash of data verified.
> Compressed 2048 bytes to 23...
> Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 1028.0 kbit/s)...
> Hash of data verified.
> Compressed 427488 bytes to 308909...
> Wrote 427488 bytes (308909 compressed) at 0x00002000 in 27.4 seconds (effective 124.6 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...

instead of `Hard resetting via RTS pin` I have un- and replugged the device, but I guess that doesn't matter

thanks for helping me! kind regards, Markus

RAZRu, 2019-12-09:
Shelly 2.5 is demanding for power supply, maybe not enough current if you try to pop web UI when it powered by USB-TTL adapter

For me personnaly algorithm looks like this:
1. Flash
2. Try to join wifi AP
3. If webUI not popping, click "Forget this AP" in wifi setting
4. Recconect again

RavenSystem, 2019-12-10:
I personally always connect Shelly devices to 230VAC after flashing.

Shimakuchi, 2019-12-10:
After connection the device to a 230VAC it worked fine. thx!

closed 2019-12-10 by Shimakuchi

jvhtec, 2019-12-30:
Hi, i'm having the same problem, flashed LCM onto my device (Generic ESP breaker), now LCM network shows up but no web configuration screen appears, tried on PC (Firefox, Chrome), Mac (Safari) and IOS to no avail, web page never appears

Shimakuchi, 2020-01-16:
Hi, flashed now around 14 devices. Every now and then I have this "issue".

1. make sure you are power your device over a 220-230VAC (or the US equivalent ~120V) as suggest above.
2. try to un- and plug again. Sometimes this helped (at least for me)
3. start over. UNPLUG the 230V and flash the device again.

hope this helps