RavenSystem/esp-homekit-devices #649: Sonoff SV as Garage Door Template

kokos1405, 2019-12-09:
Has anyone managed to setup sonoff sv as garage door?
Is there a ready template?

mientki, 2019-12-09:
I have waited a long time for this functionality :)  I also ask for an example Template for garage door opener.

noobydp, 2019-12-09:
👍

RavenSystem, 2019-12-09:
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-with-homekit-garage-door-adaptation-working-time-18-seconds-only-close-sensor-connected-to-gpio-14-and-stop-step-dry-relay-connected-to-action-garage-door-button

macjl, 2019-12-10:
Hello,

I've set up pretty similar Json. But can you explain this part?

`"2":{  
            "r":[  
               {  
                  "g":12,
                  "i":1.5
               },
               {  
                  "g":12,
                  "v":1,
                  "i":0.5
               },
               {  
                  "g":12,
                  "v":1,
                  "i":2
               }
            ]
         }`

I think it's for making two pushes on relay, but how does it work?

lfiocco, 2019-12-12:
Something does not work with the SV JSON you have on the template page. HAA keeps going into setup mode. I loaded a simple switch, it find. Garage Door, no..

noobydp, 2019-12-12:
> Something does not work with the SV JSON you have on the template page. HAA keeps going into setup mode. I loaded a simple switch, it works, it's not the device.

The sample json worked for me.

RavenSystem, 2019-12-12:
@lfiocco fixed.

RavenSystem, 2019-12-12:
@macjl 
v=0 i=1.5 relay is off and will on after 1.5 secs.

lfiocco, 2019-12-12:
Thank you working great

closed 2019-12-12 by RavenSystem