RavenSystem/esp-homekit-devices #650: Device doesn't connect to network

tonysprenk, 2019-12-09:
Hi,

I have flashed an esp8266 and I can connect to the LCM-XXXX wifi network but I've tried a few times to put in the credentials as suggested in the installation tutorial but nothing happens every time. I have tried with my iPhone and with my Mac. I have also flashed again. Any suggestions? 

Thanks A lot!

Shimakuchi, 2019-12-09:
just an idea, have you tried another browser? [Google Chrome](https://www.google.com/intl/de_de/chrome/) or [Firefox](https://www.mozilla.org/de/firefox/new/)

tonysprenk, 2019-12-09:
Haven't tried other browsers but just tried again and got some different behaviour. The Device has connected to my wifi network but when I go to its IP address in my browser it doesn't show the HAA setup screen but the same initial setup screen where you choose the OTA repository. It's also still broadcasting the LCM-XXXX SSID. So weirdly it has connected to my home wifi (I can see it under connected devices in my routers config) but it's also broadcasting its own network. I know it's the same device because the Mac address shown in my router config corresponds with LCM-XXXX said that is being broadcasted... Hopefully I'm making sense. Thanks in any case for taking the time in assisting me.

RavenSystem, 2019-12-10:
- Erase and flash it again.
- Run that command in a macOS or Linux Terminal: `nc -kulnw0 45678`
- Connect to LCM-XXXXXX wifi AP and fill all required fields, then click Join button.
- Copy and paste logs in a TXT file and attach it here.

tonysprenk, 2019-12-10:
Thanks for the suggestion. For some reason the `nc -kulnw0 45678` command doesn't output anything. I'm not familiar with the command so maybe I'm doing something wrong. Should it be outputting information in the terminal window? I'm on macOS Catalina BTW.

RavenSystem, 2019-12-10:
If there is nothing, then device is not connected to your wifi. What command are you using to flash device?

tonysprenk, 2019-12-10:
I used Tuya-Convert to flash the device. Maybe something went wrong there, but I have managed to flash many devices in the same way.

RavenSystem, 2019-12-10:
Sorry, I don't know how Tuya-Convert works. But problem seems to be a bad flash size: It must be fixed 1MB, and maybe Tuya-Convert uses autodetect. Better ask at Tuya-Convert repo.

tonysprenk, 2019-12-10:
Yes I think that may be the problem. These devices we’re flashed with an old version of Tuya-Convert. Unfortunately these devices are bricked I think...

RavenSystem, 2019-12-10:
If device is bricked, you can open and try soldering pines to flash with wires.

tonysprenk, 2019-12-10:
Thanks for all the help!

closed 2019-12-10 by tonysprenk

Rocket200, 2020-01-01:
> Hi,
>
> I have flashed an esp8266 and I can connect to the LCM-XXXX wifi network but I've tried a few times to put in the credentials as suggested in the installation tutorial but nothing happens every [...]

I Have same Problem.
From Tasmota to HAA with the FW from @kueblc